OpenRoberta/robertalab #627: cannot download in webview

tcurdt, 2017-09-27:
We have OpenRoberta running within a webview on iOS (`WKWebView`) and we would like to intercept and receive the downloads inside the app. While most other editors provide a normal or base64 encoded URL for the file, OpenRoberta provides a URL in the form of `blob:https://...` which is not something we can download from inside the app.

ensonic, 2017-09-27:
If the code is open-source, getting a link would be useful...
What download are you trying to intercept - exporting the project? Also when saying 'most other editors' - do you have an example?

tcurdt, 2017-09-27:
@ensonic sorry, no link just yet. On every navigation action a webview will get a callback with the request. If it's a hex file the navigation can be canceled and the download can/should happen outside of the webview.

```
    func webView(_ webView: WKWebView, decidePolicyFor navigationAction: WKNavigationAction, decisionHandler: @escaping (WKNavigationActionPolicy) -> Void) {

        if let download = editor.download(navigationAction.request) {

            // it's a hex download, cancel the navigation action
            decisionHandler(.cancel)

            // now download the url that has the hex file
            print(download.url.absoluteString)
```
The [microbit editor](https://makecode.microbit.org/) sends the data as base64 encoded URL.
The [calliope mini editor](https://miniedit.calliope.cc/) sends the data as a direct download.

With OpenRoberta I am getting an URL to a blob that (I assume) is internal to the webview and I don't see a way to get access to the data.

bjost2s, 2017-09-27:
I easily could remove the blob, but how can I test it?
As we are close to a release I can only commit this in a separate branch. Could you set up a server on your own and test it?

tcurdt, 2017-09-27:
@bjost2s I haven't looked into the details of the deployment procedures but I am pretty certain I should be able to test it if it is on a branch.

bjost2s, 2017-10-02:
I have checked the actual version from our lab again. Blobs only appear when downloading xml files. For Calliope you get a modal with the following html:
`<div id="downloadLink"><div id="programLink" style="text-align: center;"><br><a style="font-size:36px; padding: 20px" download="NEPOprog.hex" href="data:Application/octet-stream;content-disposition:attachment;charset=utf-8,%3A020 ... ">NEPOprog</a>`
where ... is the full hex content.

What exactly should I improve

tcurdt, 2017-10-02:
Please see the screencast:

[image]

I can see that the "NEPOprog" link is of the structure you are referring to and it should/would work - but of course one would want to check the box and just use the "run" button.

BTW: The two last points of the instructions don't really make much sense to me yet. "left column"? "save button"?

bjost2s, 2017-10-02:
Our instructions are for desktop browser. If you right click on a Link you will have the possibility To choose in the left column the Drive an the save the program there . The user doesn' t have To leave the Lab. 
What happens in your screencast, if you simply click/touch on the link. The download process should start immediatly, without asking for drives etc. I think that would be the point where the wepapp Could grap the hex.

tcurdt, 2017-10-02:
@bjost2s I didn't understand that the instruction were meant for the file save dialog :)

If I just click on the link it works fine. But as I said before: I want to check the box and just use the "run" button - and that's where things break. Because then I am just getting an url to a blob.

bjost2s, 2017-10-03:
I pushed a possible solution to branch calliopeDownload. Please check it.

bjost2s, 2017-10-03:
Sorry, forgot to remove the alert ;-), but most interesting thing is if your app could grap the hex now

bjost2s, 2017-10-09:
Any comments? Is it working in general?
I would then start some test to release it as soon as possible.

tcurdt, 2017-10-09:
@bjost2s from the diff it looks OK (at least on the first glance - given that I don't know the code base)

Unfortunately I will have to make time to work out the installation first to test this. Which means setting up a vm and the installing it. Or is there by any chance a docker image that should work?

So far I found https://hub.docker.com/r/exmatrikulator/robertalab/

bjost2s, 2017-10-10:
what we have seen and tested, it will not work. I'll send you a PM

tcurdt, 2017-10-10:
Docker: I am very certain that it can be done - but if you don't have it working it's a moo point for now.

Branch: Thanks for access to the server. I can confirm it worked for us!

Looking forward to see this merged into the next release. Thanks!

bjost2s, 2017-10-11:
seems to work on ubuntu chrome and FF, windows 10 chrome and FF, safari?

bjost2s, 2017-12-02:
It seems that you managed to grap the downloaded hex from the "blob", if so please close this, I will not merge it to develop.

tcurdt, 2017-12-02:
@bjost2s wait - what? I confirmed the branch is working and we are waiting for a release with it.

bjost2s, 2017-12-03:
sorry, someone told me that your new app is already working with the open roberta lab. 
I will merge it to develop

tcurdt, 2017-12-03:
@bjost2s Ah, sorry for the confusion. Yes, we released the app but Open Roberta is hardly usable without the change. We have another editor that works - but it would be really great to get this in so Open Roberta can also be fully supported. We just couldn't wait any longer for the release. Thanks!

bjost2s, 2018-01-25:
We have trouble with Edge and IE, so I add a fix for this browsers.

bjost2s, 2018-01-30:
please check in published 2.5.0 version online. We had to do some changes, no idea if the fix is still working for IOS webview. Please reopen, when it is still an issue. (release is already published, online tomorrow morning)

bjost2s, 2018-02-15:
not working as reported by callliope

commit 15eb2a2d (2018-02-15): fix #627 for Ipad/Iphone only
closed 2018-02-15 by bjost2s