OpenRoberta/robertalab #628: EV3 pop-up massage after connect/disconnect

Thorlogy, 2017-09-28:
After re-connecting the EV3 via USB a pop-up appears: 
Das Open Roberta Lab ist zur Zeit nicht erreichbar. Bitte überprüfe deine Internet Verbindung und lade diese Webseite neu. Falls das Problem weiterhin besteht, schreibe bitte eine Mail an folgende Adresse: roberta-zentrale@iais.fraunhofer.de 

This message is missleading (wrong), e.g. website reload is not necessary.

bjost2s, 2017-09-28:
This only appears on windows and while using the usb program with ev3!
Problem is, that windows switches the network for a short time. There is currently no way to filter it out, because it is correct: The client computer thinks for a short time, that it is not connected to the internet anymore. The message is correct, but windows cannot distinguish.

Thorlogy, 2017-09-29:
I would like to recommend to change the massage as following:

Die Internet-Verbindung wurde unterbrochen. Solltest du gerade deinen Roboter per USB Verbunden haben, genügt ein Klick auf die Programmieroberfläche. 

Falls das Problem weiterhin besteht, überprüfe deine Internet-Verbindung und lade diese Webseite neu. Sollte das Problem erneut auftreten, schreibe bitte eine E-Mail an: roberta-zentrale@iais.fraunhofer.de

closed 2017-10-02 by cvejoski