OpenRoberta/robertalab #642: NAO Face Detection Error

nicolaipoehner, 2017-10-04:
Dear all,

I finally got the OpenRobertaLab working with our NAO but found some small mistakes while playing around with it this morning:

Robot: Error!	ALFaceDetection::setRecoginitionEnabled
	Can't find method: setRecoginitionEnabled (resolved to '(b)')

I guess the mistake might be the missspelling of recognition.

JanisM91, 2017-10-04:
typo fixed with  OpenRoberta/robertalab-naoprogram@f2210ad2e8dde817cd46dcef339b2d9006ece650

bjost2s, 2017-10-05:
This fix forces us to have again a new release for the naoprogram. Wouldn't it better to fix it in the codegeneration?

bjost2s, 2017-10-09:
Is this resolved? If so please add the label resolved and move it to the right project. Is there a new release for the NAO program?

cvejoski, 2017-10-14:
fixed with ddb90cc9d615beda602fe205d2bcbc8ca975fde9

closed 2017-11-16 by bjost2s