OpenRoberta/robertalab #643: NAO Mark Detection Error

nicolaipoehner, 2017-10-04:
Dear all,

I finally got the OpenRobertaLab working with our NAO but found some small mistakes while playing around with it this morning:

Robot: [[1507108853, 407227], [[[1, -0.05625927075743675, 0.10536150634288788, 0.12622003257274628, 0.12622003257274628, 6], [64]]], [0.03334982320666313, 0.012278732843697071, 0.20545527338981628, 7.450580596923828e-09, -0.30123841762542725, 0.3527779281139374], [0.0061920415610075, 0.00459809135645628, 0.5383654832839966, -0.0019768117927014828, -0.37755340337753296, 0.3569805324077606], 'CameraTop']

I wanted NAO to detect NAO mark nr 64. I guess he does recognize it since the 64 is in that array of numbers somewhere. However I suppose output should just be the number 64 itself.

bjost2s, 2017-10-09:
Is this resolved? If so please add the label resolved and move it to the right project. Accidentially it is already released, but there was no one who checked it.

bjost2s, 2017-10-12:
It seems that now only a NAO Mark id is returned. If NAO detects more than one NAO mark which one is returned? 
I think the solution from the last commit will break if no mark is detected because of null pointer exeption. 
How can the user can get information about a detected NAO Marks location. If I want the NAO to go towards this mark, I need at least the heading values or more

closed 2017-10-26 by bjost2s