OpenRoberta/robertalab #645: Version 2.3.0: Programm import dosen't work

Dittse, 2017-10-04:
You can't upload your as XML-file saved programs anymore. I've already tried EV3, NXT and Calliope with the same Error:
[image]

paulbaumgarten-isl, 2017-10-04:
As per my comments on issue 647, this is an issue for me and my students as well.

Both the "my programs" method and "import program" create the "Attention: Server error" message as per the above screenshot.

bjost2s, 2017-10-04:
@paulbaumgarten-isl please delete the browser's cache, e.g. with Strg + F5. 
so far only old programs importing is a problem. We will fix it ASAP

paulbaumgarten-isl, 2017-10-04:
@bjost2s Thanks for the browser tip. That appears to work for any programs that were "saved" to the open-roberta cloud. But any attempt to use Import Program, even on one that I made today, does not work.

Ironically, I had been recommending to my students to use the export/import strategy rather than save to the open-roberta cloud as I'm waiting for our IT department to get our own instance of Open Roberta Server up and running and figured that would ease the transfer of student projects when that happened. The upshot being most of my students will not be able to load their current projects without the "import program" function.

paulbaumgarten-isl, 2017-10-05:
@bjost2s Update: Clearing the browser is not successful with programs saved to the open-roberta cloud either. It "appears" to load my programs in the sense that the blockly blocks appear, but as soon as I attempt to execute the program, the "server error" message appears and the program disappears.

Video screen recording demonstrating what I mean... https://drive.google.com/file/d/0BzjRDhdvMhVAX0xFWEgtVU15X28/view?usp=sharing

This means neither cloud saving nor import/export of programs is working.

nilleholger, 2017-10-05:
It still won't work. if i try to open any old program of the roberta-cloud it will crash by showing in the simulator! Afterwards also new programs won't work in the simulator.

Opening the simulator after restart will work...

cvejoski, 2017-10-06:
The fix is not yet online, probably on Monday we will have small release

bjost2s, 2017-10-07:
@paulbaumgarten-isl , @nilleholger we published a fixed version right now. Please give it a try and please don't forget to delete the browsers cache again.

paulbaumgarten-isl, 2017-10-09:
@bjost2s Tested and confirmed as fixed, thank you!
That said, I took the issues of last week as incentive to finally build our own O.R. server, which I might add was a very easy to follow process (very clear instructions). That way, we have control over the timing of upgrades and the ability to test them internally rather than discovering issues by way of 20 twelve year olds complain their robots don't work. 
This isn't a criticism - OR is providing a wonderful tool for education (I really don't like the default Lego software), and at zero financial cost to schools. I leave the comment here only as the advice I wish I could have given myself back in June and as a recommendation to other teachers evaluating OR.
Thank you.

closed 2017-10-09 by bjost2s