OpenRoberta/robertalab #647: EV3 over USB not working since site upgrade

paulbaumgarten-isl, 2017-10-04:
Hello
I run classes with school students using Open Roberta and it has been a fantastic tool (much better than the default Lego programming application). Every kit is an EV3 with the micro SD card, using USB cable to connect to Open Roberta using your USB mini-app.

Today (first lesson since the website change), no students were able to run their programs. When they attempt to execute, the EV3 screen briefly displays "NERO" (for a fraction of a second instead of the usual ~10 seconds) then returns to the Open Roberta main menu. 

Additionally (I think I see there are already tickets about this): Opening of saved programs AND importing of previously exported programs are also not working. Students had to re-write their programs from blanks.

Advice/suggestions much appreciated.

cvejoski, 2017-10-04:
When you connect the EV3 with the lab it should appear a pop up asking you to update the EV3. You should update the ev3 before running a program

paulbaumgarten-isl, 2017-10-04:
Ahhh... we had been ignoring that for the last month so I didn't think of it with this week's change. (The one time I tried it, it seemed to get stuck). Will try and report back in a couple of hours (in a class now).

paulbaumgarten-isl, 2017-10-04:
lab.open-roberta.org pop up box: "Attention: An error has occurred while updating the new firmware on your robot."

cvejoski, 2017-10-04:
What OS do you use?

paulbaumgarten-isl, 2017-10-04:
I'm using Windows 10 with Chrome. The SD card firmware is 2.0.0, and the OpenRobertaUSBSetupEN was 2_0_3.
Should I just flash my micro SD cards with the version currently on github 2.3.0, will that achieve the same thing as the over-the-wire firmware update?

cvejoski, 2017-10-04:
Yes, you can flash. It will have the same end effect

paulbaumgarten-isl, 2017-10-04:
Ok, flashing the SD cards now enabled programs to execute, thank you! (now I just have to flash the remaining 24 cards!)

I still can't load existing programs from either "my programs" or "import program" (and students have been using a mix of both strategies to save their progress). Both methods produce the "Attention: Server-error" message identified in issue 645 so I'll close this issue and add a note to that issue.

closed 2017-10-04 by paulbaumgarten-isl

cvejoski, 2017-10-04:
You should be able to update the SD cards when you connect with a token.

paulbaumgarten-isl, 2017-10-05:
If, by connecting via token, you mean using the Open Roberta USB connection app, I tried that. As shown above, I get an error "Attention: An error has occurred while updating the new firmware on your robot." It doesn't really matter, it won't take me long to manually flash the cards. Thanks anyway.