OpenRoberta/robertalab #652: Type script for the Open Roberta Lab

bjost2s, 2017-10-09:
We would like to use the advantages of static checking and code refactoring when developing out client side. Type script seems to do this job very well.
ToDo

- [ ] check what tools developers need to install to use typescript
- [ ] check if typescript can help us to get rid of the cached files problem
- [ ] make a plan how we can switch with little effort and in best case not to loose git history

philippmaurer, 2018-01-24:
The TypeScript implementation will be done in multiple steps, that do not just aim on implemention TypeScript itself, but also to rewrite the code in a way, that we can use the best parts of the tools we need to implement for the TypeScript usage anyway.

Overview:
- [ ] Environment refactoring
- [ ] Basic TypeScript Implementation
- [ ] In-depth TypeScript Implementation

## Environment refactoring
The current JavaScript environment is not ready to use TypeScript from scrath due to browser compability. To achieve that a build proess is necessary.
Having a build process enabled completely new possibilities on the way. Currently the JS code is neither sent minified, nor is it packed in an optimal file structure. This can be achieved by a build tool.
This first step will be therefore focus on the the build process, which will not contain (but consider) TypeScript compilation, but optimizations for the current code instead. In addition to that the build process will be made optional for users that just use this repository to host their own server or work on non-JavaScript fiels only, **by commiting the built files as well.**

## Basic TypeScript Implementation
The next step will be to create the TypeScript compilation process and transfer the JavaScript files to TypeScript. Upon doing this it is planned to use the TypeScript fundamental strength, but not to alter the codes logic. The main aim will be to find type related bugs and weaknesses in the code, that need to be fixed immediately, but skip improvements and optimizations to speed up the implementation process.

## In-depth TypeScript Implementation
At last the code will be reworked to make the most use of the TypeScript functionailites to optimize and improve the code.

rehmsen, 2018-06-25:
We are doing a hackathon at Google Munich today and tomorrow and I'd be interested to see if there is anything I can help with on the Webpack/TypeScript side. I checked out the `webpack` branch, and it seems to not have been touched much since February. Do you have any update of what the status is here, what you are blocked on and what it would take to finish webpack support and get it merged back into `develop`?

rehmsen, 2018-06-25:
Quick update: I rebased the existing webpack branch onto develop and fixed the merge conflicts and a couple compile errors. It mostly seemed to work already. I then added source maps and changed the `pom.xml` to install node/npm/webpack and run it, to not inconvenience non-front-end developers with this change. Frontend devs can run `webpack --watch` and get the auto rebuild when changing input files to not regress on the edit-refresh experience. I am looking to clean it up some more and get input from the core contributors before sending a pull request. Here is my current code:

https://github.com/rehmsen/robertalab/tree/webpack2

rehmsen, 2018-06-26:
This is the state of my fork right now:
- [x] Integrate with Maven such that it installs and runs everything needed as part fo the regular build - this should alleviate the need to check in compiled sources, which tends to be a mess to keep in sync.
- [x] Output and serve sourcemaps
- [x] Fix translation module loading
- [x] Put bundled JS into `target/` instead of into the source tree
- [ ] Fix dynamic loading of robot files in the simulator
- [ ] Move JS out of `staticResources`, because they do not need to be statically served anymore
- [ ] Document somewhere how to run `webpack --watch`
- [ ] Setup live reload functionality

closed 2019-06-26 by rbudde