OpenRoberta/robertalab #664: The cookie disclaimer is only shown, when the first-start-modal is shown aswell

philippmaurer, 2017-10-18:
Currently the cookie disclaimer is part of the first-start-modal. Since that modal is only shown on specific pages, the disclaimer should be separated from the first-start-modal.

philippmaurer, 2017-10-20:
Testcases:

Testcase 1: The user gets the possibility to accept the storage of cookies on his/her computer:
1.) Open the lab for the first time
2.) A disclaimer appears on the bottom of the page, providing the user general information about cookies and its usage on the open-roberta lab. The disclaimer also provides a possibility for the user to accept the given information. The text indicates clearly, that by accepting the given statement, cookies will be saved.

Testcase 2: The user can accept the storage of cookies on his/her computer: 
1.) Open the lab for the first time
2.) A disclaimer appears on the bottom of the page.
3.) The disclaimer contains a button, that cleary indicates, that by activating it the given statement is accepted.
4.) Click or tab on the accept-button.
5.) The disclaimer disappears.

Testcase 3: The user can close the cookie-disclaimer without accepting the storage of cookies:
1.) Open the lab for the first time
2.) A disclaimer appears on the bottom of the page.
3.) The disclaimer contains a button, which cleary indicates, that by activating it the disclaimer will disappear and the given statement is not accepted.
4.) Click or tab on the deny-button.
5.) The disclaimer disappears.

Testcase 4: The cookie disclaimer does not disappear, if the user does not ineract with it
1.) Open the lab for the first time
2.) A disclaimer appears on the bottom of the page.
3.) Do something on the lab page, preferably something that opens a modal popup. Under no circumstances interact with the cookie disclaimer. This may make some features inaccessable, but thats ok.
4.) The cookie disclaimer is always visible and never fully, nor partially, hidden behind another element of the page.
5.) The cookie disclaimer is still visible upon closing the page

Testcase 5: No cookies outlasts the session time, if the cookie statement was not accepted:
1.) Open the lab for the first time
2.) A disclaimer appears on the bottom of the page.
3.a.) Either do not accept the cookie statement by not interacting with the cookie disclaimer
3.b.) Or close the cookie disclaimer via the deny-button.
4.) Do something on the lab page. If you have the possiblity to make a decision that sticks with you, even after leaving the page, do it. For example select a robot in the first-start-popup and activate, that your desicion will be remembered.
5.) Check, if any cookies were stored. (Developer Tools -> Application/Storage/Debugger -> Cookies)
6.) If so, leave the page. Check again, if any cookies are still saved. (In your OS cookie folder, or hard-reload the page and check, if the cookies are still present with the same values)
No cookies should be stored.

Testcase 6: Upon accpeting the cookie statement cookies are stored:
1.) Open the lab for the first time
2.) A disclaimer appears on the bottom of the page.
3.) Accept the cookie statement by activating the accept-button.
4.) Do something on the lab page. If you have the possiblity to make a decision that sticks with you, even after leaving the page, do it. For example select a robot in the first-start-popup and activate, that your desicion will be remembered.
5.) Check, if any cookies were stored. (Developer Tools -> Application/Storage/Debugger -> Cookies)
6.) If so, leave the page. Check again, if any cookies are still saved. (In your OS cookie folder, or hard-reload the page and check, if the cookies are still present with the same values)
The cookies should be stored.

Testcase 7: Cookies are saved, even if the cookie statement was accepted after the cookie was stored:
1.) Open the lab for the first time
2.) A disclaimer appears on the bottom of the page.
4.) Do something on the lab page. If you have the possiblity to make a decision that sticks with you, even after leaving the page, do it. For example select a robot in the first-start-popup and activate, that your desicion will be remembered.
3.) Accept the cookie statement by activating the accept-button.
5.) Check, if all neccessary cookies were stored. (Developer Tools -> Application/Storage/Debugger -> Cookies) 
The cookie-names should be most likely identical with those from testcase 6.

cvejoski, 2017-10-24:
Clicking on **show details** forwards to non-existing page https://roberta-home.de/index.php?id=137&L=1.

I assume this will be fixed with the new home page?

closed 2017-10-24 by cvejoski