OpenRoberta/robertalab #667: Add Chinese localization

et2010, 2017-10-20:
Hi,

Sorry for the duplicated issue: https://github.com/OpenRoberta/blockly/issues/27.

I really want to help since I want to teach my 10 year to learn Linux and programming. Kudos for this great project.

bjost2s, 2017-11-25:
We are working on a better translation process. As soon as it is available we will invite you to translate ;-) (maybe next week)
Thank you

mpasternak, 2017-12-20:
@bjost2s , how are things? I'd love to improve translation in Polish for the website. Where should I look for in the repo?

bjost2s, 2017-12-20:
I will send you the invitation tomorrow, but I need to be clear about your gmail adress and languages you would like to translate, could you please send me the information to beate.jost  at  iais.fraunhofer.de?
Thank you

philippmaurer, 2017-12-21:
@et2010 The remodelling of the translation process is finished and we would be happy to invite you into our translater circle. We use a Google Drive sheet for the translation management. To add you to it you need a google account. 
Can you please send me a mail with your google email address to my gmail adress (displayed in my profile), so we can add you to it?

Best regards
Philipp

et2010, 2017-12-22:
Hi, Philipp

I'm really glad to join you. I've sent you an email. Thanks!

bjost2s, 2018-03-12:
will be continued outside this project.

closed 2018-03-12 by bjost2s