OpenRoberta/robertalab #669: Sugerencia Bloque posición (Suggestion Position Block)

adelmar1971, 2017-10-23:
Bloque posición X, Y
Que permita en el simulador, ubicar al robot en una coordenada X,Y
Esto amplía las posibilidades de simulación en proyectos específicos.
------------
X, Y position block
That allows in the simulator, locate the robot in a coordinate X, Y
This extends the possibilities of simulation in specific projects.

ensonic, 2017-10-23:
Unfortunately such a block would not work for the real robots as we have them. It would require a positioning system (e.g. SLAM) for the robot.

adelmar1971, 2017-10-26:
Saludos. Comprendo la idea. Pero... lo que planteo es la posibilidad de posicionar el robot (sólo en el simulador). Estoy claro que en la vida real, no es fácil; pero en el simulador, este bloque permitiría automatizar más algunos procesos que puede aprovecharse para promover compentencias en línea.
[Greetings. I understand the idea. But ... what I propose is the possibility of positioning the robot (only in the simulator). I am clear that in real life, it is not easy; but in the simulator, this block would allow to automate some more processes that can be used to promote online championships.]

closed 2019-06-26 by rbudde