OpenRoberta/robertalab #670: Posibilidad de guardar escenario [Scenery save possibility]

adelmar1971, 2017-10-23:
Sugerencia:
Que los escenarios subidos como imagen, puedan guardarse y estar asociados al proyecto.
----------
Suggestion:
That the scenarios uploaded as image, can be saved and be associated with the project.

bjost2s, 2017-10-23:
Currently we do not have projects but programms that can be saved. As you can share programs with everybody we have to be very careful and we feel responsible what is shared. If we allow to share images we have to moderate the lab. Unfortunately this is not possible. Your uploaded image is stored in the local storage, so if you come back to the lab, please step through the list of  simulation backgrounds, your custom one is the last one.

adelmar1971, 2017-10-26:
## Saludos. Es cierto. Se comprende el criterio de seguridad. Aprobado.
Greetings. It is true. The criterion is understood. Approved.

closed 2017-10-26 by adelmar1971