OpenRoberta/robertalab #673: Lego RCX integrattion?

adelmar1971, 2017-10-26:
### ## Es posible incorporar la controladora Lego RCX? En Venezuela, y de seguro en otros países, hay un buen número de instituciones educativas que aún la utilizan.

Is it possible to integrattion the Lego RCX controller? In Venezuela, and certainly in other countries, there are a number of educational institutions that still use it.

bjost2s, 2017-10-26:
It would be possible and I would like to see it here (also have an RCX at home ;-)), but the effort would be very high because of the outdated kind of connection. If we could find a solution, then only for RCX USB tower.
I will put it to the wish list and maybe someone would like to investigate in it.

bjost2s, 2017-11-12:
@ensonic interested to work on this ;-) ?

ensonic, 2017-11-12:
I've found this german post about how to use it under linux:
https://trollheaven.wordpress.com/2017/05/14/lego-mindstorms-unter-linux/
I'd don't think I have enough time for it though. I think I have one such brick, but no idea if it still works.

adelmar1971, 2017-11-12:
Greetings @bjost2s @ensonic 
I do not have the technical knowledge to do what is required, but I do need it. We have more than 30 RCX in the school and there are at least 55 schools that still use RCX in Caracas, Venezuela. Surely there are more in the world.
Some reference page:
https://github.com/Glitchbone/mindstorms-rcx-osx-tools
The RCX works with the NQC language and with the infrared tower in two options: USB port or Serial port.
I am ready to test. I will look for students interested in the project in the university. Thank you very much.
-----------
Saludos @bjost2s @ensonic 
No tengo el conocimiento técnico para hacer lo requerido, pero sí la necesidad. Tenemos más de 30 RCX en el colegio y hay al menos unas 55 escuelas que todavía utilizan RCX en Caracas, Venezuela. De seguro hay más en el mundo.
Alguna página de referencia: 
https://github.com/Glitchbone/mindstorms-rcx-osx-tools
El RCX funciona con lenguaje NQC y con la torre infraroja en dos opciones: puerto USB o puerto Serial.
Estoy a la orden para hacer pruebas. Buscaré en la universidad estudiantes interesados en el proyecto. Muchas gracias.

rbudde, 2019-06-26:
at the moment not feasable

closed 2019-06-26 by rbudde