OpenRoberta/robertalab #674: Indicate where blocks modify the input (e.g. clip max speed)

ensonic, 2017-10-27:
Assume a block like:
```
[ Drive forward speed % ] [ 5000 ]
```

Since the speed value is to be given in percent, we will limit the input to a value from 0...100. It would be nice to indicate the to the user. We could do this with a checker. Ideally we put the warning sign into the number box though.

bjost2s, 2017-11-12:
For lejos and nxt etc, there is no need to limit it. What exactly is the problem of executing speed 5000?

ensonic, 2017-11-12:
Speed is in % where 100 % is the maximum speed. If the motor could go twice as fast, then twice as fast would be 100 % :) If we don't clip the values for lejos/nxt, then the lejos libraries probably do this, right?