OpenRoberta/robertalab #675: Help users to understand large numbers (e.g. beep for 7000000000 ms)

ensonic, 2017-10-27:
I notices that some kids like to use large numbers. A common case where this can be problematic is the duration on the beep block. I even heard the justification "it is just milliseconds". It would be nice to show the kids how much the value is actually in years, month, days, hours, ...

This could be done:
* as a comment in the generated code
* as a dynamic comment on the block itself (needs to be refreshed when the input changes)

The checker is probably not the right place, since we don't want to add a hard limit.

rishabhdeepsingh, 2018-03-02:
For this, we need to set an upper range that after that range the dialog box will appear. What should be the upper range 1 Hr or 1 day or something else?

rbudde, 2019-06-26:
I'd prefer a hint and an example in the help.