OpenRoberta/robertalab #680: Improve message handling

aBrAEUMER, 2017-11-02:
Currently the `send message` and `receive message` blocks are pausing the running program until the message is sent or received successfully (on my EV3 with ev3dev at least - and I even can not abort it by pressing "enter" and "down" on the brick).
This makes using messages quite frustrating or complex or useless in many cases (as explained later).

Therefore it would be nice have messages handled like a sensor, especially have a
`repeat ... until message received`
or
`repeat ... until message from connection ... received`
block.

Another way to improve the message handling would be a possibility to store the incoming messages in a List/Queue for futher treatment without having to wait/pause until a new messages is received.

Both solutions seem to require the message receiver run in a additional thread on the brick IMHO and may be complex to implement.
Improving the sending of a message may be easier to implement, e.g. by setting a timeout value (maybe specified by the user as a parameter in the block) and adding a boolean `last message send successfully` block. 

My initial problem was to have multiple EV3s search for an item (red ball) in a room. All robots should stop searching if one robot found the item.
At the current state of NEPO the robots can not wait for the "found it"- message from another robot while still processing the search loop. Therefore all robots have to send "still searching"- messages all the time, which makes the protocol quite complex to design and handle by our pupils - and there is a lot of overhead to process, too. Finally, there is more waiting to recieve/send messages than searching for the item and this makes the current message handling useless for many problem statements, I think.

ensonic, 2017-11-02:
Good points. I'll definitely fix the message blocks on ev3dev to be interruptible.

patrickschaerer, 2017-11-19:
I tried to write a program with bluetooth messages today and realized that the text messages are received as shown:
sent: connected
received (or printed on the core): b'connected'
I tried to use message sending and receiving to program a remote control, but I could not get it working.
the if did not recognize the right "connected"
Mabe you could add this as well to your improvements.
[NEPOprog.xml.zip](https://github.com/OpenRoberta/robertalab/files/1485755/NEPOprog.xml.zip)

aBrAEUMER, 2017-11-19:
The issue you described has been fixed by ensonic about two weeks ago. See 
[this closed issue](https://github.com/OpenRoberta/robertalab-ev3dev/issues/51). It is a matter of how python 3 handles UTF-8 coded strings. You have to replace the ev3.py on your brick [by the latest ev3.py](https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/ev3.py) and restart your ev3. 

.

patrickschaerer, 2017-11-19:
> You have to replace the ev3.py on your brick by the latest ev3.py and restart your ev3.

How do I do that? Where is the file ev3.py located?

aBrAEUMER, 2017-11-19:
I will write a little howto tomorrow because I do not have a ev3 right now. It should be very simple using a linux system where you can mount the SD card. I did it directly on the brick using the midnight commander and a private webspace reachable via ftp.
The location of ev3.py can be found easily by typing 
`find / -name ev3.py`
in the ssh. There are two files of this name, but only one in a openroberta folder (the other one is in a ev3dev folder).

patrickschaerer, 2017-11-20:
ok I am new to linux but I'll try
Thank you

ensonic, 2017-11-20:
I'll soon do a release, so that you can just update the package.

patrickschaerer, 2017-11-20:
Thank you very much. I found the files and replaced them. Now the code is working.

patrickschaerer, 2017-11-20:
Vielen Dank!
Ich konnte das File finden und ersetzen.
Liebe Grüsse
Patrick Schärer

ensonic, 2017-11-20:
For the message blocks, one thing we should do is to improve the docs and actually state how the blocks behave. Right now the docs don't say anything about whether the receiveMassage would return the empty string if nothing got send.

rbudde, 2019-06-26:
I believe, that a satisfying solution is possible only, when we add parallel execution to Nepo. We're discussing this. I'll close this issue, but the request is remembered :-)

closed 2019-06-26 by rbudde