OpenRoberta/robertalab #685: Run simulation from code directly?

DanTulovsky, 2017-11-08:
Hi.

Is it possible to run the simulator from code directly?  I am at a point where I want to write in code, rather than using blocks, but would love to be able to run the code through the simulator.  Is there an easy way to accomplish this?

Thank you
dan

DanTulovsky, 2017-11-08:
This is for ev3dev/python specifically.

ensonic, 2017-11-09:
We don't let the user edit the code right now, since there is no mapping of arbitrary code back to blockly. The simulator is based on the code structure defined by the blocks, so there is currently no way to run this for arbitrary code.

bjost2s, 2017-11-12:
For the simulation we cannot use python (or java). Instead we generate lists of commands evaluated by javascript. So no way to use python or any other programming language. We also do not have a plan to support this.

closed 2017-11-12 by bjost2s