OpenRoberta/robertalab #692: Functions renaming themselves

Dittse, 2017-11-10:
If you place these three blocks in a certan order (from top to bottom) and open the simulation, the called function "doSomething" will rename itself to "doSomething2":

1. 
[image]

2.
[image]

3.
[image]

This behavior only occurs under certain circumstances:
- the default name of a function in the currently selected language mustn't have any spaces (so only german and english are concerned)
- the base function (2.) has to have the default name of a function of that language
- the other function (3.) has to start with that default name and continues with a attachment of any kind (in this example the number "2")
- the blocks are placed in one of the following orders: 
 -132
 -312
 -231

The genarted code seems unremarkable.

elhele, 2017-11-13:
It also changes the name if try to upload the program, so it is not only simulation issue.

elhele, 2017-11-13:
It is a purely blockly bug that was already fixed. In a blockly repo a year ago it was also broken in the same way. So, to fix it we need to update blockly.

bjost2s, 2017-11-14:
it is a blockly bug, but we have created our own renaming process. So we will fix it in our way, so no need to update to blockly which we definitly can't do for the next release ( and don't want to)

commit c4b15ce0 (2017-12-28): add fix #692, #718 from blockly
closed 2018-01-12 by VinArt