OpenRoberta/robertalab #694: Sound sensor implementation for EV3

boonto, 2017-11-13:
Implement sound sensor blocks for EV3.

bjost2s, 2017-11-22:
I have moved this issue to TODO, because I couldn't find an implementation for ev3dev. 
Please also implement it there.

ensonic, 2017-11-23:
The soundSensor should be already supported:
https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/roberta/ev3.py#L140
I need to find one to test it though and check the the code-generator makes the blocks.

ensonic, 2017-12-02:
The sound-sensor block reads "get value sound sensor port 1", should this be "get level sound sensor port 1". Also in German we could call it "Gib Lautstärke Geräuschsensor Port 1".

boonto, 2017-12-15:
Now it says "get sound % sound sensor" and "Gib Geräusch % Geräuschsensor" in german which doesnt sound right. I think value/Wert or level/Pegel are both better.

The ev3dev implementation currently returns negative dB values, can you map them to the 0-100 range?

ensonic, 2017-12-15:
> The ev3dev implementation currently returns negative dB values, can you map them to the 0-100 range?
With 0 for quiet and 100 for loudest? The db-scale is kind of open ended.

boonto, 2017-12-15:
I think for children its more intuitive the other way round: 0 being silent and 100 being loud.

closed 2018-01-15 by bjost2s