OpenRoberta/robertalab #704: Building OpenRoberta in Windows 10 (Error)

DJuego, 2017-11-21:
Environment:

- Windows 10 x64
- jdk-8u152-windows-x64
- apache-maven-3.5.2

```
git clone git://github.com/OpenRoberta/robertalab.git
cd robertalab/OpenRobertaParent
mvn clean install

```

Output:

```
[INFO]
[INFO] --- jacoco-maven-plugin:0.7.5.201505241946:report (report) @ RobotEV3 ---
[INFO] Analyzed bundle 'RobotEV3' with 44 classes
[INFO]
[INFO] --- maven-jar-plugin:2.4:jar (default-jar) @ RobotEV3 ---
[INFO] Building jar: X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\target\RobotEV3-2.4.0-SNAPSHOT.jar
[INFO]
[INFO] --- maven-dependency-plugin:2.9:copy (copyUpdateResourcesLejos0.9.0) @ RobotEV3 ---
[INFO] Configured Artifact: de.fhg.iais.openroberta.ev3.v0:EV3Runtime:2.3.0:jar
[INFO] Configured Artifact: de.fhg.iais.openroberta.ev3.v0:EV3Menu:2.3.0:jar
[INFO] Configured Artifact: org.json:json:20140107:jar
[INFO] Configured Artifact: org.java-websocket:Java-WebSocket:1.3.0:jar
[INFO] Copying EV3Runtime-2.3.0.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v0\EV3Runtime.jar
[INFO] Copying EV3Menu-2.3.0.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v0\EV3Menu.jar
[INFO] Copying json-20140107.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v0\json.jar
[INFO] Copying Java-WebSocket-1.3.0.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v0\Java-WebSocket.jar
[INFO]
[INFO] --- maven-dependency-plugin:2.9:copy (copyUpdateResourcesLejos0.9.1) @ RobotEV3 ---
[INFO] Configured Artifact: de.fhg.iais.openroberta.ev3.v1:EV3Runtime:2.4.0:jar
[INFO] Configured Artifact: de.fhg.iais.openroberta.ev3.v1:EV3Menu:2.4.0:jar
[INFO] Configured Artifact: org.json:json:20140107:jar
[INFO] Configured Artifact: org.java-websocket:Java-WebSocket:1.3.0:jar
[INFO] Copying EV3Runtime-2.4.0.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v1\EV3Runtime.jar
[INFO] Copying EV3Menu-2.4.0.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v1\EV3Menu.jar
[INFO] Copying json-20140107.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v1\json.jar
[INFO] Copying Java-WebSocket-1.3.0.jar to X:\OpenRoberta\robertalab\OpenRobertaParent\RobotEV3\resources\updateResources\lejos_v1\Java-WebSocket.jar
[INFO]
[INFO] --- maven-dependency-plugin:2.9:copy (copyCrossCompilerResourcesLejos0.9.1) @ RobotEV3 ---
[INFO] Configured Artifact: de.fhg.iais.openroberta.ev3.v1:EV3Runtime:2.4.0:jar
[INFO] Configured Artifact: org.json:json:20140107:jar
[INFO] Configured Artifact: de.fhg.iais.roberta:ev3classes:0.9.1-beta:jar
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO]
[INFO] RobertaParent ...................................... SUCCESS [  1.018 s]
[INFO] OpenRobertaRobot ................................... SUCCESS [ 21.041 s]
[INFO] RobotEV3 ........................................... FAILURE [ 10.516 s]
[INFO] RobotNXT ........................................... SKIPPED
[INFO] RobotArdu .......................................... SKIPPED
[INFO] RobotNAO ........................................... SKIPPED
[INFO] RobotMbed .......................................... SKIPPED
[INFO] OpenRobertaServer .................................. SKIPPED
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time: 32.729 s
[INFO] Finished at: 2017-11-21T18:35:34Z
[INFO] Final Memory: 45M/739M
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-dependency-plugin:2.9:copy (copyCrossCompilerResourcesLejos0.9.1) on project RobotEV3: Unable to find artifact.: Failure to find de.fhg.iais.roberta:ev3classes:jar:0.9.1-beta in http://maven.iais.fraunhofer.de/artifactory/openroberta-release was cached in the local repository, resolution will not be reattempted until the update interval of ORA-releases has elapsed or updates are forced
[ERROR]
[ERROR] Try downloading the file manually from the project website.
[ERROR]
[ERROR] Then, install it using the command:
[ERROR]     mvn install:install-file -DgroupId=de.fhg.iais.roberta -DartifactId=ev3classes -Dversion=0.9.1-beta -Dpackaging=jar -Dfile=/path/to/file
[ERROR]
[ERROR] Alternatively, if you host your own repository you can deploy the file there:
[ERROR]     mvn deploy:deploy-file -DgroupId=de.fhg.iais.roberta -DartifactId=ev3classes -Dversion=0.9.1-beta -Dpackaging=jar -Dfile=/path/to/file -Durl=[url] -DrepositoryId=[id]
[ERROR]
[ERROR]
[ERROR]   de.fhg.iais.roberta:ev3classes:jar:0.9.1-beta
[ERROR]
[ERROR] from the specified remote repositories:
[ERROR]   ORA-releases (http://maven.iais.fraunhofer.de/artifactory/openroberta-release, releases=true, snapshots=false),
[ERROR]   ORA-snapshots (http://maven.iais.fraunhofer.de/artifactory/openroberta-snapshot-local, releases=false, snapshots=true),
[ERROR]   central (https://repo.maven.apache.org/maven2, releases=true, snapshots=false)
[ERROR] -> [Help 1]
[ERROR]
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR]
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR]
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :RobotEV3
```

labcode-de, 2017-11-21:
Which branch do you use? master or develop?

DJuego, 2017-11-21:
**master**...
i can try the develop branch.

DJuego

DJuego, 2017-11-21:
Well. I have build successfully the **develop** branch (no problems)
However:
```
> sh ./ora.sh --start-from-git
2017-11-21 20:08:59,004 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - default properties from classpath. Using the resource: openRoberta.properties
2017-11-21 20:08:59,008 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - new property from command line: database.mode=embedded
2017-11-21 20:08:59,008 [main] session-id=[] user-id=[] robot-name=[] INFO de.fhg.iais.roberta.util.Util1 - new property from command line: database.parentdir=OpenRobertaParent/OpenRobertaServer
2017-11-21 20:08:59,009 [main] session-id=[] user-id=[] robot-name=[] INFO d.f.i.roberta.util.RobertaProperties - As temporary directory C:\Users\YO\AppData\Local\Temp\ will be used

```
And when i try: http://localhost:1999/
 ERR_CONNECTION_REFUSED

DJuego
>

VinArt, 2017-11-21:
I think, that `database.parentdir=OpenRobertaParent/OpenRobertaServer` can be a problem, try putting full path there. Don't forget to copy dbbase to db-2.4.0.

cvejoski, 2017-11-21:
We have change the location of the maven artifactory. We will have a new version in master with update maven artifactory URL.

DJuego, 2017-11-21:
(**develop** branch)

I created a  robertalab\OpenRobertaParent\OpenRobertaServer\db-2.4.0 dir
and i copied the content of robertalab\OpenRobertaParent\OpenRobertaServer\dbBase dir to db-2.4.0 dir.

It works!!

Thank you very much, @VinArt 

I will wait the new version in master, @cvejoski . Thanks! 

This Saturday, 25  i will have a show with Open Roberta but. I think i can use the develop branch for my purposes.

DJuego

bjost2s, 2017-11-22:
master just got a fix, tomorrow we will create a new release out of it (no more changes for that). so please use master, to be sure.
@VinArt, please do not close serious issues before they are fixed, we also have agreed in the team, who should close issues.
@DJuego, thank you very much for reporting this! We wouldn't have discovered this, because we often forget to delete the .m2 folder for testing.

bjost2s, 2017-11-22:
fixed here: c1799cc9c12e90b2dcf5ea8def8cd7009db83374

closed 2017-11-25 by cvejoski