OpenRoberta/robertalab #707: Stop button UI lacking

lotu, 2017-11-22:
The OpenRoberta UI lacks a stop button.  You can stop a program that is running but pushing <enter>+<down> but there is nothing to indicate this to the user unless they read the wiki.  Possible solutions include:

1. Turn play button into stop button when program is running
1. Add default startup text on robot (currently blank screen) stating how to stop program.

bjost2s, 2017-11-22:
Thanks, 
1. impossible, because the robot is not asking for  commands during program execution.
2. we will check, if we could provide some information during the starting period

lotu, 2017-11-23:
There would be some additional value in providing information during the starting period, this way the user see something other than a blank screen which can be confused with the robot being off.  If for some reason the robot isn't moving (loose cable) this would be extra confusing.

ensonic, 2017-11-23:
I agree that it sound like a good idea.  To add to what @bjost2s said, not all robots are connected while the program runs. The play button is "transmit + run" and on most robots the program can later be rerun from the robot without the lab being there.

When the program runs (and the play button is grayed out), we could add a tooltip to tell how to abort stop the program on the robot. Maybe we can even show a dialog with such information, if one clicks the grayed play button. Would that help?

I would not want to show this each time the program starts as this could become annoying.

lotu, 2017-11-23:
I like the tool tip idea.    I would agree that showing a dialog on the computer screen would be annoying and might not be as useful.  However, right now the display on the robot is blank and unused, so putting anything on shouldn't bother people (presuming you don't make thing click to acknowledge).  Anyone who wants to show something else would clear the screen and start displaying the other thing they want to show.

I would see this as some type of graphic showing which buttons we mean, though text would work just as well,  saying, "Program running push down + enter to stop".  This provides useful information to anyone who picks up the robot.

In the educational setting the information on how to stop the robot needs to be transmitted to each student, sometimes more than once so having it be the default thing to display means one less thing for a teacher to have to handle.

bjost2s, 2017-11-25:
Tooltips are not visible for touchscreen users. We offer in the wiki a short descrrrrbtion for each system, e.g. how to use the menu, how to start a program from the robot .... Maybe we put it there?

patric30, 2017-11-30:
A stop button would be most intuitive. Could we send an empty program to the robot to terminate execution of the one running in an endless loop or going crazy? Stop button in available in the original lego software and highly utilized by beginners. 
Most users intuitively are using the back button which kills the program & and connection which is highly frustrating. 
If stop button is not feasible I would vote for any text element to educate the user on the "enter" + "down" option to terminate. Relying on documentation is not a valid option in my mind.

ensonic, 2017-11-30:
@patric30 the robertalab does not send the program. The robot sends 'empty' requests with long timeouts to the server and when the server needs the robot to do things it 'answers' the pending request with a cmd. When the program runs, there is no pending request.

During a roberta course I switch the lab to full-screen to avoid e.g. the back button. So far I have not seen the kids using the back button.

If tooltips won't be an option, maybe we can use the status area on the top toolbar. Right now we say something like "Your program XXX will run in a moment". We could tell there how to stop it once it runs and fade that text out once the program stopped (aka the robot communicated back).

lotu, 2017-12-15:
I agree with @patric30 when I started using open Roberta on the EV3 I had trouble figuring out how to stop the program despite being a professional software developer.   I tried pushing the back button on the robot which did work but also required reconnecting.   Realistically people will not read the wiki before using OpenRobtera especially not students so this should be a obvious thing form a User Experience perspective.

I like the idea of a message in the top toolbar with how to stop instructions, I might even highlight it to make it more obvious because most people are likely to miss it.  

As an aside would it be reasonable for the robot to keep sending these requests while it was executing the program an another thread so that it can keep receiving commands.  This could be made even more sophisticated by including telemetry about the robot so it could be displayed much like what happens in the Lego software.

ensonic, 2017-12-15:
The stop button discussion is a duplicate of #218. I'll change the title of this ticket to discuss how to inform users of how to stop the program.

closed 2018-01-10 by bjost2s

lotu, 2018-02-15:
The toast helps.  Unfortunately I missed when I was playing around with the robot.  If I didn't know it existed I would not have known to look for it.

This is pattern I observed in myself and what I expect users especially young and inexperienced users would experience.  

1. Program 
2. Push run
3. Look at robot in anticipation
4. Toast informing you how to stop program appears
5. Robot starts moving
6. Toast goes away
7. Realize robot is caught in infinite loop
8. Look at screen for how to stop robot can't find answer

This is a hard UX problem to handle because you are trying to inform the user about something while the screen is not the most interesting thing they are looking at.   Looking briefly at the code submitted the simplest solution might be to extend the time the how to stop is displayed.  This would work especially well if we could dismiss the toast when the robot reconnects.