OpenRoberta/robertalab #708: Cannot Join Non Strings (But can display them)

lotu, 2017-11-22:
You can not join non string or convert numbers to strings.  This makes displaying the value of a sensor difficult.

For example this works

    show text get angle gyroscope Port 2
      in column 0
      in row 0

but this does not:

    show text Create text with "Gyro Angle: "
                               get angle gyroscope Port 2
      in column 0
      in row 0

This should be a simple matter of applying the str function to non string objects in the join.

    - hal.drawText(str(BlocklyMethods.textJoin("Hallo", hal.getGyroSensorValue('2', 'angle'))), 0, 3)
    + hal.drawText(str(BlocklyMethods.textJoin("Hallo", str(hal.getGyroSensorValue('2', 'angle')))), 0, 3)

bjost2s, 2017-11-22:
What about this?
[image]

We will discuss, if we want to have non string objects in the string category

lotu, 2017-11-23:
Yes this works though it requires you known the length of the the text.   If you wanted do something more complicated like print the gyro angle as part of an arbitrary string it would require a lot more work.  For example if you wanted to print the line, "rotations right: %d left: %d" % (get rotation encoder B, get rotation encoder C).  It is still possible to do this but much more of a pain likely requiring variables.

I agree there is some worry about the type system, but I think every type should have a string representation, otherwise it is very difficult to print.   You have two options main options allow casting or automatically coerce types.  Given the audience this Open Roberta is aimed at I would lean towards automatically coercing types, because the average teacher won't understand types themselves much less be able to explain them to students.

bjost2s, 2017-11-25:
Sorry, I think you are absolutely right. This is aready available for ev3Lejos. We will fix it in ev3dev and check it for the NAO (Python) implementation

VinArt, 2017-11-27:
This method actually does the string concatination:
```python
@staticmethod
def textJoin(*args):
    return "".join(args)
```
Should we do pre-conversion to string somehow like this:
```python
def joinTest(*args):
    return ''.join(str(arg) for arg in args)
```
?

ensonic, 2017-12-01:
I'll do this. I was about to prepare a release anyway.

closed 2018-01-30 by bjost2s