OpenRoberta/robertalab #710: Could use arm block

lotu, 2017-11-22:
It is very common for robots to have arms on them if look at the starnard EV3 robot it has a arm that turns at 1:1 with the motor but other ratios are possible.  It would be desirable to have a simple block that controlled the arm for students that don't yet have the ability to understand the abstraction of motor turns gears which turns arm.

The arm block would be like the drive block it would give the option to turn the arm up/down (or left/right, or open/closed based on config).  It would then have degrees so you could tell the arm to go up 30º this is most useful for students.

An arm block should ideally include a timeout where if the arm is blocked because it hit the ground or robot the robot doesn't just hang there.  Dealing with timeouts and problem like that are very difficult for young students.

bjost2s, 2017-11-22:
You mean the middle motor?
We always work with our standard model which has no arm. 
We can have a discussion about expanding our standard model.

lotu, 2017-11-23:
Yes the medium motor though you could use any sized motor for an arm.  Here is an example of what the arm on the standard EV3 model  https://www.youtube.com/watch?v=lrCNUchzlwc&feature=youtu.be

closed 2019-06-26 by rbudde