OpenRoberta/robertalab #712: The raspberry pi server images should configure "lab.open-roberta.org" in /etc/hosts

ensonic, 2017-11-24:
This would allow people to just use the raspberry pi as a drop-in replacement without having to change the server-address on the computers or the EV3 robots.
This is okay since the raspberry pi would not provide any connectivity to the real internet.

bjost2s, 2017-11-25:
we provide a full working raspi image. If we conigure to much, we have to provide several versions because some would like to use with internet connection. 
We could write an instruction in the readme of the raspi image, how and when to configure this feature.

Could you write the instruction here or in this wiki?

RBEGamer, 2017-11-26:
where i can find the image

bjost2s, 2017-11-27:
https://drive.google.com/open?id=0ByU4YIlmCnL3TlBpczBDNlY5UVk
Very soon it will be available on the roberta-home server here: https://www.roberta-home.de/lab/

ensonic, 2017-11-28:
> Could you write the instruction here or in this wiki?
Sure, is the Readme.txt that is part of the download in git somewhere?