OpenRoberta/robertalab #713: Disordered parameters when using microbit getPixelValue(x,y)

dr0i, 2017-11-25:
For example, doing `setPixelValue(0,2,1)` the pendant `getPixelValue(0,2,1)` wouldn't come with `1` as the result but with `0`. Changing the `x` with the `y` seems to do the trick but that's weired also the code-block says "get LED brightness x" while it's in fact `y`.

I use `lab.open-roberta.org` to code for `calliope mini`. To code for the `micro:bit` boards open-roberta uses the [https://github.com/lancaster-university/microbit-dal](microbit-dal) library. Maybe there is the bug in that library. Quick scan looked ok but I am no c++ programmer.

Here is the code-snippet of the `micro:bit` robot (which also doesn't work as expected):
```cpp
import microbit

microbit.display.set_pixel(0, 2, 1);
microbit.display.scroll(str(microbit.display.get_pixel(0, 2)))
```

VinArt, 2017-11-27:
@dr0i thank you for reporting this bug.
Affected: mbed simulation
With real board it was working, please correct me if I am wrong.

dr0i, 2017-11-28:
@VinArt ACK, real board working.

closed 2018-01-09 by bjost2s

dr0i, 2018-02-05:
I think, as 9d33010f4e04f30100feac3227cae36b9bece92d was reverted, this issue should be reopened.