OpenRoberta/robertalab #714: Write Text does not clear display

patric30, 2017-11-30:
Writing text on the display is not clearing the display first which leads to black boxes if values are changing over time. This is not intuitive for beginners.

ensonic, 2017-11-30:
This is mentioned in the docs and I'd say expected. How much should it clear? If you print "Hello" and then "I", both at (0,0) - the "I" is narrower and would only clear parts of the "H" if we'd clear the background.

patric30, 2017-12-04:
I was thinking of the (0,0) as a grid/table on the display. Whenever I write to (x,y) this cell gets cleared completely independent of the width of the old text.

bjost2s, 2018-05-07:
What about using the "clear display" block right before the "write text" block?

closed 2019-06-26 by rbudde