OpenRoberta/robertalab #717: Simulation "for each item ... in list" not working in a loop

bjost2s, 2017-12-03:
beiliegendes Programm liefert in der Simulation: true false true   false true  . Darauf folgt nichts sichtbares.
Es sollte endlos angezeigt werden: true false true   true false true ...
Die Schleife "Für Wert aus der Liste" funktioniert auch in anderen Programmen nicht.

[image]

VinArt, 2017-12-04:
Which robot is this? For Ev3 simulation is funny, it actually stops and behaviour is "true-false-true-false-true"
[image]

bjost2s, 2017-12-05:
Simulation problems are usually not related to a specific robot type, especially if the block is a not robot related one.

closed 2017-12-28 by bjost2s