OpenRoberta/robertalab #718: comment blocks would be great

jexp, 2017-12-05:
Either allowing comments on arbitrary blocks
Or adding comment blocks that are visible in the program.

Otherwise it's hard for the kids to document e.g. program usage or the "why"

bjost2s, 2017-12-05:
This feature is already on our TODO list, probably in a hidden on ;-), we try to implement it for the next release

bjost2s, 2017-12-10:
I put the new block into the text catagory, but there is maybe a better idea?

ensonic, 2017-12-18:
I am not sure people will find it there. It is really a unique feature. But I understand that a new category with a single block is overkill.

VinArt, 2017-12-29:
I wonder if it makes sense to introduce single line/multi line comments and using this block with said "mode"

bjost2s, 2017-12-29:
@VinArt please create a new ticket for your thoughts, because this one is already resolved.
Anyway I think there is no need to show multiple line comments, because you can easily add several blocks if you need it and also blockly doesn't support multiple line inputfields in the block directly. We already have multiple line comments in each block, right click and choose comment. We could think about adding this to the generated source code.

closed 2017-12-29 by bjost2s