OpenRoberta/robertalab #719: Oracle Issues

jexp, 2017-12-05:
We have a number of oracle issues when using open-roberta

One is that kids can log in but not save their programs, it fails with a "duplicate entry" error when trying to save.

Another one is this "not logged in" issue that happens several times.

The user is logged in but the system thinks they are not, see below:

[image]

cvejoski, 2017-12-05:
Do you use the public version of the OpenRoberta Lab?

ensonic, 2017-12-05:
FYI: ORA stands for OpenRobertA :)

bjost2s, 2017-12-28:
trying to fix this, but when I am logged in I can always save programs etc. Any idea how to run into this issue?

closed 2019-06-26 by rbudde