OpenRoberta/robertalab #722: Add parameters to say block

JanisM91, 2017-12-06:
Add parameters to the say block for NAO to make it possible to change the speed and the shape (pitch) of the voice.

For more information and general discussion look at #425

boonto, 2017-12-15:
We should unify the parameter ranges so they are the same on all robots. I suggest using the ranges 0-100 for speed and shape and map these values to the specific ones in the hals.

Also, I think shape should be renamed to pitch.

ensonic, 2017-12-15:
> Also, I think shape should be renamed to pitch.
+1

If we want to change the parameter ranges, lets do this before I cut another ev3dev release. I am fine with 0...100 values. For espeak the ranges are not too large:
```
-p <integer>
	   Pitch adjustment, 0 to 99, default is 50
-s <integer>
	   Speed in words per minute, 80 to 450, default is 175
```
For speed we'll have an odd default though (51.44), so we could agree to go with 50 and the docs need to be explicit and state that those are not percentages, since 0 is mapped to just a low speed.

ensonic, 2017-12-15:
@boonto or this https://github.com/kripken/speak.js - then we have the same parameters.

boonto, 2017-12-15:
The demo and helloworld examples with the prebuilt speak.js do not work. Building it myself did not work either. Next, I tried [meSpeak.js](http://www.masswerk.at/mespeak/), which adds Webkit support. However, it lead to errors in Firefox.

I believe the Web Speech API is the better approach going forward, as it supports most of the browsers out of the box and does not rely on any additional libraries.

boonto, 2017-12-18:
@ensonic in order to avoid weird mapping I think we can reduce the range of speed to 100 - 350 with default 175. The speech synthesis sounds fine until about 350, above it gets nearly unintelligible. The defaults for the block would then be 30 for speed and 50 for pitch.

closed 2018-01-14 by bjost2s