OpenRoberta/robertalab #725: ev3dev steer block does not behave correctly

boonto, 2017-12-11:
The steering block with the distance parameter has a few issues:

- If speed left = x and speed right = -x the program stops with a "division by zero" error.
- The same with left = 0 and right = 0
- If one of the speed values equals 0 the program does not stop
- In general, negative values lead to unexpected behaviour

ensonic, 2017-12-11:
I'll address with with another release. But we should probably catch more of those with the program checker.

ensonic, 2017-12-18:
We need to agree on how we'd like to handle this. Long ago I've files #122. Personally I would like to actually throw an error for bad parameter values and then use some mechanism to report those to the UI.

I'll check what the lejos hal is doing.

ensonic, 2017-12-20:
I believe negative speeds do work. But negative angle/distance might not. If you have the exact program, that'd be great.

Still what should we do for e.g. speed=0. This is a programming error. We have the syntax checker if the value is static. But if it is the result of a math block, we'll only notice at runtime. 
1. I can of course check in python and just return (so that it won't crash), but that will hide the error and the users will wonder why it is not driving
2. In some cases I'd prefer to throw an exception and return an error code to the lab (#122)

It would be nice to come up with a policy for robots hals. WDYT?

dlech, 2017-12-20:
Funny, I had the same questions/thoughts: https://github.com/ev3go/ev3dev/pull/50#issuecomment-330375362

My inclination is that you should just make the program wait forever. If you have a speed of 1, you will probably see the same behavior as well because it is not enough to turn the motor.

If you make an error, it should be that the motor is stalled in general, not that speed = 0. But, this is really more of a status indication of an abnormal condition than an error.

ensonic, 2018-01-18:
Can someone test on lejos whether is hangs if speed becomes zero?

bjost2s, 2018-01-19:
What do you mean with hang?. As far as i know, if you have one motor 0 it turns around the other one as expected, if both are 0 it doesnt't Mode. It never breaks.

ensonic, 2018-01-19:
What I meant is if distance>0 and speed==0, the block will wait for distance to be reached which never happens.

boonto, 2018-01-19:
I just tested it, the method finishes immediately.
As far as I can see this is not specified in the Roberta ev3lejosv0/v1 runtimes and probably implicit in [DifferentialPilot (v0)](http://www.lejos.org/nxt/pc/api/lejos/robotics/navigation/DifferentialPilot.html#setTravelSpeed(double))/[MovePilot (v1)](http://www.lejos.org/ev3/docs/lejos/robotics/navigation/MovePilot.html#travel-double-).

ensonic, 2018-01-19:
I'll let you guys close it after testing.

closed 2018-12-14 by ensonic