OpenRoberta/robertalab #726: use filesaver.js for calliope + microbit

ensonic, 2017-12-12:
There is a library called [filesaver.js](https://github.com/eligrey/FileSaver.js/) that lets you trigger save-as from links. [demo](https://eligrey.com/demos/FileSaver.js/). 
We could use this to make the downloading less painful, especially on laptops with a touch-pad the right-click is not easy to do.

bjost2s, 2017-12-14:
for chrome on linux and windows 10 the demo does nothing then downloading direkt to the download folder.

ensonic, 2017-12-15:
It looked like it works when trying on my chromebook since I had set the download folder to MINI: and the calliope was not conencted and this way I've got the save_as dialog.
Filed a bug for the library: https://github.com/eligrey/FileSaver.js/issues/390

closed 2019-06-26 by rbudde