OpenRoberta/robertalab #729: Wiki article needs information on starting service openrobertalab

bittner, 2017-12-30:
The Wiki article "[EV3dev Robot and WiFi](https://jira.iais.fraunhofer.de/wiki/display/ORInfo/EV3dev+Robot+and+WiFi)" is incomplete.

Between step 1 and step 2 the service `openrobertalab` needs to be enabled as follows:

1. Connect to the LEGO brick using SSH: ([default password](http://www.ev3dev.org/docs/tutorials/connecting-to-ev3dev-with-ssh/) is `maker`)
```bash
ssh robot@ev3dev.local
```
2. On the brick run:
```bash
sudo systemctl unmask openrobertalab
sudo systemctl start openrobertalab
```
From that on the `openrobertalab` service will automatically be started each time the brick boots.

Background Information
-----------------
@dlech [explained on Gitter today](https://gitter.im/ev3dev/chat?at=5a47d430edd2230811f460cb) why this is necessary.

bjost2s, 2018-02-01:
Our users, especially the wiki users have no idea how to set up an ssh connection. Usually they  work with a windows system and need detailed step by step instructions. Also for all articles that we have in our wiki, we got a lot of mails where people are asking for help.
Wouldn't it be better to explain all this stuff on the ev3dev pages so that we can link to your sites in general?

ensonic, 2018-02-01:
I have documentation here:
https://github.com/OpenRoberta/robertalab-ev3dev/blob/develop/README.md

I'll do a change later to add the ssh-info. But ideally we can indeed enable this form the UI.

ensonic, 2018-02-01:
@bittner thanks for the suggestions. Saw the gitter chat with dlech. I am usually there too and happy to discuss the options.

bittner, 2018-02-01:
You're probably referring to my [declaration of intent](https://gitter.im/ev3dev/chat?at=5a47eba429ec6ac3119e9055) to add the feature of starting (any) service to the UI? Well, sooner or later - if no-one else does it ... I may get that going. :smirk:

closed 2019-06-26 by rbudde