OpenRoberta/robertalab #730: Bug in Generating Text

LukePink, 2018-01-02:
The Value of a DistanceSensor is unable to be shown in a text; it will apear as 'undefined'.

Intention: Debugging ,y program, which depends on the distance based on UltrasonicSensor.

Method: Simulation, Standart Config of EV3basis.

Detailes:
I try to create program, that races to a collition, and just bevore collition slow down, to approach slower. And finaly before colootion it should stop.
PREUDOCODE:
until (Ultrasound says close)
    drive vorward fast
    Display Distance on Display
until (Ultrasound says imideatly infront of s.th.)
    drive vorward slow
    Display Distance on Display
stop

See:
[https://lab.open-roberta.org/#loadProgram&&ev3lejosv1&&DriveToCollition_2&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22ev3%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22737%22%20y=%22-1274%22%3E%3Cblock%20type=%22robProcedures_defreturn%22%20id=%22h2w1/4=%7C=2%5Bwn*q!b0)a%22%20intask=%22true%22%3E%3Cmutation%20declare=%22true%22%20return_type=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22NAME%22%3EcloseToSomething%3C/field%3E%3Cfield%20name=%22TYPE%22%3EBoolean%3C/field%3E%3Crepetitions%3E%3Cstatement%20name=%22ST%22%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22T*i%25dKa0ltpFON=%5D=A%7B@%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22false%22%20declaration_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3EminDist%3C/field%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3C/block%3E%3C/statement%3E%3Cstatement%20name=%22STACK%22%3E%3Cblock%20type=%22robProcedures_ifreturn%22%20id=%22-%7Co=w::sE)ty6d/6%5Dq-U%22%20intask=%22true%22%3E%3Cmutation%20value=%221%22%20return_type=%22Boolean%22%3E%3C/mutation%3E%3Cvalue%20name=%22CONDITION%22%3E%3Cblock%20type=%22logic_compare%22%20id=%22?yI2%5DDQw!=1%7B-gdmk0%7B.%22%20intask=%22true%22%3E%3Cfield%20name=%22OP%22%3ELT%3C/field%3E%3Cvalue%20name=%22A%22%3E%3Cblock%20type=%22robSensors_ultrasonic_getSample%22%20id=%22@/@dameCS/TFak0tRpY#%22%20intask=%22true%22%3E%3Cmutation%20mode=%22DISTANCE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EDISTANCE%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3E4%3C/field%3E%3Cdata%3Eev3%3C/data%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22B%22%3E%3Cblock%20type=%22variables_get%22%20id=%22*X%7D;mDY.(w@1D7*XHgRu%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3EminDist%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22logic_boolean%22%20id=%22q%7D1MBa%5Ba%7Cmt%25FqnFj%5Bez%22%20intask=%22true%22%3E%3Cfield%20name=%22BOOL%22%3ETRUE%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/statement%3E%3Cvalue%20name=%22RETURN%22%3E%3Cblock%20type=%22logic_boolean%22%20id=%22~ET7QjG4m*-%7Di%7CCqV:%5DZ%22%20intask=%22true%22%3E%3Cfield%20name=%22BOOL%22%3EFALSE%3C/field%3E%3C/block%3E%3C/value%3E%3C/repetitions%3E%3C/block%3E%3C/instance%3E%3Cinstance%20x=%22736%22%20y=%22-1078%22%3E%3Cblock%20type=%22robProcedures_defreturn%22%20id=%22WyTgjalo5i~(4w#@:%25!;%22%20intask=%22true%22%3E%3Cmutation%20declare=%22true%22%20return_type=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22NAME%22%3EdriveCloseTo%3C/field%3E%3Cfield%20name=%22TYPE%22%3EBoolean%3C/field%3E%3Crepetitions%3E%3Cstatement%20name=%22ST%22%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22FjWCQSo194ReJqP%25uA!1%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22true%22%20declaration_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Edistance%3C/field%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3C/block%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22q%7C-lQe2y%5E9W%5BtS%5E%5ENii!%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22true%22%20declaration_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Es_%3C/field%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3C/block%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22%5Dg;ykcb9u?WzFK%60L%5BBC%60%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22true%22%20declaration_type=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Et_%3C/field%3E%3Cfield%20name=%22TYPE%22%3EBoolean%3C/field%3E%3C/block%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22f2b%5DxpbiYuO)z.,*cUT%5D%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22false%22%20declaration_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Ecounter%3C/field%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3C/block%3E%3C/statement%3E%3Cstatement%20name=%22STACK%22%3E%3Cblock%20type=%22variables_set%22%20id=%22E;.Ou5is-1+B%7BRfR0~_G%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Et_%3C/field%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22robSensors_touch_isPressed%22%20id=%2229Gdde9Dq=#%5BGnszD4RD%22%20intask=%22true%22%3E%3Cfield%20name=%22SENSORPORT%22%3E1%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3Cblock%20type=%22controls_whileUntil%22%20id=%221G84StV5!%7DC+6W=@00e%60%22%20intask=%22true%22%3E%3Cfield%20name=%22MODE%22%3EWHILE%3C/field%3E%3Cvalue%20name=%22BOOL%22%3E%3Cblock%20type=%22logic_operation%22%20id=%22@ZMHs3lgk5lfSW.ICx=w%22%20intask=%22true%22%3E%3Cfield%20name=%22OP%22%3EAND%3C/field%3E%3Cvalue%20name=%22A%22%3E%3Cblock%20type=%22logic_negate%22%20id=%22AEcI%60%5Br%5D)Q)m7Z%25/0Ph7%22%20intask=%22true%22%3E%3Cvalue%20name=%22BOOL%22%3E%3Cblock%20type=%22variables_get%22%20id=%22qhlmiGsiuN=e+s+A6+sN%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Et_%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22B%22%3E%3Cblock%20type=%22logic_negate%22%20id=%22%7CA3Kst%7D#%5Dy/y%7D2=pc%7B:0%22%20intask=%22true%22%3E%3Cvalue%20name=%22BOOL%22%3E%3Cblock%20type=%22robProcedures_callreturn%22%20id=%22j2m?/%5D9%7D@IXJ6y;rNyuS%22%20intask=%22true%22%3E%3Cmutation%20name=%22closeToSomething%22%20output_type=%22Boolean%22%3E%3Carg%20name=%22minDist%22%20type=%22Number%22%3E%3C/arg%3E%3C/mutation%3E%3Cvalue%20name=%22ARG0%22%3E%3Cblock%20type=%22variables_get%22%20id=%22qBz%7BTu#NxJ@5x:N~i@bM%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Edistance%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3Cstatement%20name=%22DO%22%3E%3Cblock%20type=%22robActions_display_text%22%20id=%22;%5ENlrH@fCNItdxq47%7CdU%22%20intask=%22true%22%3E%3Cvalue%20name=%22OUT%22%3E%3Cblock%20type=%22text%22%20id=%22p.1XtaK%5D1fq1wGmy)n8s%22%20intask=%22true%22%3E%3Cfield%20name=%22TEXT%22%3EThe%20distance%20is:%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22COL%22%3E%3Cblock%20type=%22math_number%22%20id=%22WfXjnEoRGiqaN)ykleH8%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E0%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ROW%22%3E%3Cblock%20type=%22math_number%22%20id=%22z4BGvKcw4bZyCb%7B+f?%7BZ%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E0%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3Cblock%20type=%22robActions_display_text%22%20id=%22?D;qd5qRLWuR*oK!%7D=B%5B%22%20intask=%22true%22%3E%3Cvalue%20name=%22OUT%22%3E%3Cblock%20type=%22robText_join%22%20id=%2277%5DrExw(Pf(u,K;nRQdT%22%20intask=%22true%22%3E%3Cmutation%20items=%225%22%3E%3C/mutation%3E%3Cvalue%20name=%22ADD0%22%3E%3Cblock%20type=%22text%22%20id=%22SsJxe_F/%5BT%60PiwRo%5Bb0o%22%20intask=%22true%22%3E%3Cfield%20name=%22TEXT%22%3E'%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD1%22%3E%3Cblock%20type=%22variables_get%22%20id=%22-Ts.07jmvaiajzwXXd*:%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Ecounter%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD2%22%3E%3Cblock%20type=%22text%22%20id=%22)9Z.2d~%7BT72m%7BAH#*fc9%22%20intask=%22true%22%3E%3Cfield%20name=%22TEXT%22%3E'%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ADD3%22%3E%3Cblock%20type=%22robSensors_ultrasonic_getSample%22%20id=%22Ky.%7D)kQI=rHUCWB;A(rL%22%20intask=%22true%22%3E%3Cmutation%20mode=%22DISTANCE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EDISTANCE%3C/f [... cut ...] 22%3EOFF%3C/field%3E%20%3Cfield%20name=%22MOTOR_DRIVE%22%3ERIGHT%3C/field%3E%20%3C/block%3E%20%3C/value%3E%20%3Cvalue%20name=%22MC%22%3E%20%3Cblock%20type=%22robBrick_motor_big%22%20id=%227%22%20intask=%22true%22%3E%20%3Cfield%20name=%22MOTOR_REGULATION%22%3ETRUE%3C/field%3E%20%3Cfield%20name=%22MOTOR_REVERSE%22%3EOFF%3C/field%3E%20%3Cfield%20name=%22MOTOR_DRIVE%22%3ELEFT%3C/field%3E%20%3C/block%3E%20%3C/value%3E%20%3C/block%3E%20%3C/instance%3E%20%3C/block_set%3E%3C/config%3E%3C/export%3E](url)

bjost2s, 2018-01-02:
Please provide a minimal example and tell us the firmware you are using, ev3dev, ev3 lejos 0.9.1, ev3 lejos 0.9.0 or the simulation. 
For me your program does nothing than beeping once (in the simulation), no movement and 'undefined' on the display in the simulation.
If you want to make it run before we fix the missing casting in the text concatenation block, please try this form of generation output on the display:
https://lab.open-roberta.org/#loadProgram&&ev3lejosv1&&NEPOprog&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22ev3%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22-163%22%20y=%22-196%22%3E%3Cblock%20type=%22robControls_start%22%20id=%22fkUYOXd%7DV1uZigQokbK*%22%20intask=%22true%22%20deletable=%22false%22%3E%3Cmutation%20declare=%22false%22%3E%3C/mutation%3E%3Cfield%20name=%22DEBUG%22%3ETRUE%3C/field%3E%3C/block%3E%3Cblock%20type=%22robControls_loopForever%22%20id=%22qbpcZHR%5B%5EMF;Pbmh%60#Cm%22%20intask=%22true%22%3E%3Cstatement%20name=%22DO%22%3E%3Cblock%20type=%22robActions_display_text%22%20id=%22YHdOT*eOOxttm8cXX%7BT2%22%20intask=%22true%22%3E%3Cvalue%20name=%22OUT%22%3E%3Cblock%20type=%22text%22%20id=%226;m6cBD(5UQ7=EF16?66%22%20intask=%22true%22%3E%3Cfield%20name=%22TEXT%22%3EHi%20ultrasonic%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22COL%22%3E%3Cblock%20type=%22math_number%22%20id=%22XJRHpF9%7DaH%7B%7Cp%60:m@iRR%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E0%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ROW%22%3E%3Cblock%20type=%22math_number%22%20id=%22VrFZb%5D#%7C%7DwHFndfttNYg%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E0%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3Cblock%20type=%22robActions_display_text%22%20id=%22SZ%5EW.Z/JHeoI*)-vr%257%7B%22%20intask=%22true%22%3E%3Cvalue%20name=%22OUT%22%3E%3Cblock%20type=%22robSensors_ultrasonic_getSample%22%20id=%22kRGb28_SO5%7D!t/6#%7D%7BE=%22%20intask=%22true%22%3E%3Cmutation%20mode=%22DISTANCE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EDISTANCE%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3E4%3C/field%3E%3Cdata%3Eev3%3C/data%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22COL%22%3E%3Cblock%20type=%22math_number%22%20id=%22HEB:X*EVM%7D*-%60%7DwDfRpt%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E12%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22ROW%22%3E%3Cblock%20type=%22math_number%22%20id=%22I!#O*_KQo8d%25sPv%5D%5B399%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E0%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3Cblock%20type=%22robControls_wait_time%22%20id=%22Buzitq5sy#1a;fqkB20m%22%20intask=%22true%22%3E%3Cvalue%20name=%22WAIT%22%3E%3Cblock%20type=%22math_number%22%20id=%22?5:Gad;H;T%60tYaU=kpj%5B%22%20intask=%22true%22%3E%3Cfield%20name=%22NUM%22%3E500%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3Cblock%20type=%22robActions_display_clear%22%20id=%225cOoi/N0o%5D~0c(d%7CxT8e%22%20intask=%22true%22%3E%3C/block%3E%3C/statement%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/program%3E%3Cconfig%3E%3Cblock_set%20robottype=%22ev3%22%20xmlversion=%22%22%20description=%22%22%20tags=%22%22%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cinstance%20x=%22213%22%20y=%22213%22%3E%3Cblock%20type=%22robBrick_EV3-Brick%22%20id=%221%22%20intask=%22true%22%20deletable=%22false%22%3E%3Cfield%20name=%22WHEEL_DIAMETER%22%3E5.6%3C/field%3E%3Cfield%20name=%22TRACK_WIDTH%22%3E18%3C/field%3E%3Cvalue%20name=%22S1%22%3E%3Cblock%20type=%22robBrick_touch%22%20id=%222%22%20intask=%22true%22/%3E%3C/value%3E%3Cvalue%20name=%22S2%22%3E%3Cblock%20type=%22robBrick_gyro%22%20id=%223%22%20intask=%22true%22/%3E%3C/value%3E%3Cvalue%20name=%22S3%22%3E%3Cblock%20type=%22robBrick_colour%22%20id=%224%22%20intask=%22true%22/%3E%3C/value%3E%3Cvalue%20name=%22S4%22%3E%3Cblock%20type=%22robBrick_ultrasonic%22%20id=%225%22%20intask=%22true%22/%3E%3C/value%3E%3Cvalue%20name=%22MB%22%3E%3Cblock%20type=%22robBrick_motor_big%22%20id=%226%22%20intask=%22true%22%3E%3Cfield%20name=%22MOTOR_REGULATION%22%3ETRUE%3C/field%3E%3Cfield%20name=%22MOTOR_REVERSE%22%3EOFF%3C/field%3E%3Cfield%20name=%22MOTOR_DRIVE%22%3ERIGHT%3C/field%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22MC%22%3E%3Cblock%20type=%22robBrick_motor_big%22%20id=%227%22%20intask=%22true%22%3E%3Cfield%20name=%22MOTOR_REGULATION%22%3ETRUE%3C/field%3E%3Cfield%20name=%22MOTOR_REVERSE%22%3EOFF%3C/field%3E%3Cfield%20name=%22MOTOR_DRIVE%22%3ELEFT%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/instance%3E%3C/block_set%3E%3C/config%3E%3C/export%3E

LukePink, 2018-01-03:
Thank you for the update with the generation of text. This WorkAround does seem to do the trick.

I use the simulation provided (in Area "drawing scene") with Firefox 57.0.3 (64 Bit) on Linux. I made sure no content is blocked by AdBlocker e.g.

When I run the simulation with the (updated) programm, It beeps, start rolling with a green light ( as is expected) untill the distance information from the Ultrasonic sensor says '<=19' (also as expected). Afterwards it drives vorward with slower speed and an orange light (again as expected). But id does not stop when the sensor says '<=10' as is defined.

And: I am sorry, but I was unable to reproduce the behaviour with an other program. But the Sound and light can be deleted, and still bring the same result. 
So the "smaller program" is:
[https://lab.open-roberta.org/#loadProgram&&ev3lejosv1&&DriveToCollition_2&&%3Cexport%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%3E%3Cprogram%3E%3Cblock_set%20xmlns=%22http://de.fhg.iais.roberta.blockly%22%20robottype=%22ev3%22%20xmlversion=%222.0%22%20description=%22%22%20tags=%22%22%3E%3Cinstance%20x=%22674%22%20y=%22-1267%22%3E%3Cblock%20type=%22robProcedures_defreturn%22%20id=%22h2w1/4=%7C=2%5Bwn*q!b0)a%22%20intask=%22true%22%3E%3Cmutation%20declare=%22true%22%20return_type=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22NAME%22%3EcloseToSomething%3C/field%3E%3Cfield%20name=%22TYPE%22%3EBoolean%3C/field%3E%3Crepetitions%3E%3Cstatement%20name=%22ST%22%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22T*i%25dKa0ltpFON=%5D=A%7B@%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22false%22%20declaration_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3EminDist%3C/field%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3C/block%3E%3C/statement%3E%3Cstatement%20name=%22STACK%22%3E%3Cblock%20type=%22robProcedures_ifreturn%22%20id=%22-%7Co=w::sE)ty6d/6%5Dq-U%22%20intask=%22true%22%3E%3Cmutation%20value=%221%22%20return_type=%22Boolean%22%3E%3C/mutation%3E%3Cvalue%20name=%22CONDITION%22%3E%3Cblock%20type=%22logic_compare%22%20id=%22?yI2%5DDQw!=1%7B-gdmk0%7B.%22%20intask=%22true%22%3E%3Cfield%20name=%22OP%22%3ELT%3C/field%3E%3Cvalue%20name=%22A%22%3E%3Cblock%20type=%22robSensors_ultrasonic_getSample%22%20id=%22@/@dameCS/TFak0tRpY#%22%20intask=%22true%22%3E%3Cmutation%20mode=%22DISTANCE%22%3E%3C/mutation%3E%3Cfield%20name=%22MODE%22%3EDISTANCE%3C/field%3E%3Cfield%20name=%22SENSORPORT%22%3E4%3C/field%3E%3Cdata%3Eev3%3C/data%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22B%22%3E%3Cblock%20type=%22variables_get%22%20id=%22*X%7D;mDY.(w@1D7*XHgRu%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3EminDist%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/value%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22logic_boolean%22%20id=%22q%7D1MBa%5Ba%7Cmt%25FqnFj%5Bez%22%20intask=%22true%22%3E%3Cfield%20name=%22BOOL%22%3ETRUE%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3C/statement%3E%3Cvalue%20name=%22RETURN%22%3E%3Cblock%20type=%22logic_boolean%22%20id=%22~ET7QjG4m*-%7Di%7CCqV:%5DZ%22%20intask=%22true%22%3E%3Cfield%20name=%22BOOL%22%3EFALSE%3C/field%3E%3C/block%3E%3C/value%3E%3C/repetitions%3E%3C/block%3E%3C/instance%3E%3Cinstance%20x=%22668%22%20y=%22-1074%22%3E%3Cblock%20type=%22robProcedures_defreturn%22%20id=%22WyTgjalo5i~(4w#@:%25!;%22%20intask=%22true%22%3E%3Cmutation%20declare=%22true%22%20return_type=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22NAME%22%3EdriveClose%3C/field%3E%3Cfield%20name=%22TYPE%22%3EBoolean%3C/field%3E%3Crepetitions%3E%3Cstatement%20name=%22ST%22%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22FjWCQSo194ReJqP%25uA!1%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22true%22%20declaration_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Edistance%3C/field%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3C/block%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22q%7C-lQe2y%5E9W%5BtS%5E%5ENii!%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22true%22%20declaration_type=%22Number%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3Espeed2%3C/field%3E%3Cfield%20name=%22TYPE%22%3ENumber%3C/field%3E%3C/block%3E%3Cblock%20type=%22robLocalVariables_declare%22%20id=%22%5Dg;ykcb9u?WzFK%60L%5BBC%60%22%20intask=%22true%22%20deletable=%22false%22%20movable=%22false%22%3E%3Cmutation%20next=%22true%22%20declaration_type=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3EtouchedSomeThing%3C/field%3E%3Cfield%20name=%22TYPE%22%3EBoolean%3C/field%3E%3C/block%3E%3C/statement%3E%3Cstatement%20name=%22STACK%22%3E%3Cblock%20type=%22variables_set%22%20id=%22:dTxq_/G_isUP-/xTg:k%22%20intask=%22true%22%3E%3Cmutation%20datatype=%22Boolean%22%3E%3C/mutation%3E%3Cfield%20name=%22VAR%22%3EtouchedSomeThing%3C/field%3E%3Cvalue%20name=%22VALUE%22%3E%3Cblock%20type=%22logic_boolean%22%20id=%220*l9sFIrky6c.uR#OG*p%22%20intask=%22true%22%3E%3Cfield%20name=%22BOOL%22%3EFALSE%3C/field%3E%3C/block%3E%3C/value%3E%3C/block%3E%3Cblock%20type=%22controls_whileUntil%22%20id=%221G84StV5!%7DC+6W=@00e%60%22%20intask=%22true%22%3E%3Cfield%20name=%22MODE%22%3EWHILE%3C/field%3E%3Cvalue%20name=%22BOOL%22%3E%3Cblock%20type=%22logic_operation%22%20id=%22@ZMHs3lgk5lfSW.ICx=w%22%20intask=%22true%22%3E%3Cfield%20name=%22OP%22%3EAND%3C/field%3E%3Cvalue%20name=%22A%22%3E%3Cblock%20type=%22logic_negate%22%20id=%22A [... cut ...] 22%3EOFF%3C/field%3E%20%3Cfield%20name=%22MOTOR_DRIVE%22%3ERIGHT%3C/field%3E%20%3C/block%3E%20%3C/value%3E%20%3Cvalue%20name=%22MC%22%3E%20%3Cblock%20type=%22robBrick_motor_big%22%20id=%227%22%20intask=%22true%22%3E%20%3Cfield%20name=%22MOTOR_REGULATION%22%3ETRUE%3C/field%3E%20%3Cfield%20name=%22MOTOR_REVERSE%22%3EOFF%3C/field%3E%20%3Cfield%20name=%22MOTOR_DRIVE%22%3ELEFT%3C/field%3E%20%3C/block%3E%20%3C/value%3E%20%3C/block%3E%20%3C/instance%3E%20%3C/block_set%3E%3C/config%3E%3C/export%3E](url)

closed 2018-01-15 by elhele