OpenRoberta/robertalab #731: LogIn/LogOut Bug on opening new page

LukePink, 2018-01-02:
When I am logged in in the Tab A, and open (inadvertantly or not) a new Tab B with [https://lab.open-roberta.org/](url), The user in A is apperantly logged out, as it is assumed the User ob B is logged in; which is no one.

The buggy thing is: the User in A does not notice this until an action is performed where A has to be loggen in, on which he/she will be informed " ORA_USER_ERROR_NOT_LOGGED_IN ". and even logging out again, results into " Invalid Javascript-command ". so one has no chanse but to reload the page. So if the Algorithm (blocks) are changed, one either loses them, or has to export them onto the Device.

bjost2s, 2018-01-02:
This behaviour comes from the sessions created on server side. If you want to be logged in twice, you can use for your second log in either another browser or an incognito tab.

LukePink, 2018-01-03:
I makred this Behaviour as a bug, not because I wanted to be logged in twice, but because the User is not being informed or asek to login again.
I was mearly pointing out, that accidental behaviour, like opening the 'help-->FAQ' in a new tab of the browser, causes the current session to be disrupted.
This also happens if one opens a "program link" in the same browser. 

I had hoped, that there would be a way to extend the current session to any new tabs of OpenRobertaLab, so the same person is logged in there. Or maybee that there is a warning, before the session is interrupted.

Stex, 2018-11-08:
I would agree that this is not the expected behaviour that users already "learned" from other sites and would generally be perceived rather as annoying than an actual feature from an UX standpoint.

Wouldn't it be possible to e.g. add a cookie "remember me" token (optionally through a corresponding checkbox) during the login process which could then be retrieved by another tab?

I don't know how the server-side session handling is implemented here, but it looks like the session id sent through the site cookie is shared between tabs (as expected), the SPA simply doesn't receive a "user is still logged in" of some kind resp. doesn't respect it.

rbudde, 2019-06-26:
will be solved when load balancing etc. is released

closed 2019-06-26 by rbudde