OpenRoberta/robertalab #735: assigning a new array cause a compiler error, calliope2017

kwollw, 2018-01-07:
[image]

Declaration and instantiation (line 11) restricts list (<array>) to 3 elements.

Assigning a new list with more elements is possible in nepo block language but will cause error in cpp (line 17).

elhele, 2018-01-15:
Now we use a library that allows assigning a new array only of the same length. We will add a possibility to assign an array of a different length in the release after the upcoming one.

bjost2s, 2018-03-19:
we should immediately switch back to array, otherwise current programs will not be compilable anymore:
@VinArt and @elhele, please make sure not to work on tickets that do not belong to a dedicated project. Here we have a good example where no review has happened, because it is not in any project. Unfortunately the solution is bad, because it will probably break other working code, because you cannot use a list in the same way like an array.

[image]

@VinArt, please undo this. We will discuss this issue further in

bjost2s, 2018-04-01:
The problem of assigning arrays in c is much more complex than expected and will be solved in general here #797 or in sub issues from #797

closed 2018-04-01 by bjost2s