OpenRoberta/robertalab #736: Global declaration is not always added in generated python code (ev3dev)

mmakai, 2018-01-08:
Hi,

Take the simple program (see screenshot):
- declare a global variable i with initial value 0
- increase its value by one with the special UI block
[image]

The generated code on ev3dev throws:
Fehler im Ev3
UnboundLocalError
local variable 'i' referenced

Likely bug: The "global i" declaration is missing from the generated code. If I set the value of i before increasing it, it works.

Generated code downloaded from roberta UI:
#!/usr/bin/python

from __future__ import absolute_import
from roberta.ev3 import Hal
from roberta.BlocklyMethods import BlocklyMethods
from ev3dev import ev3 as ev3dev
import math

class BreakOutOfALoop(Exception): pass
class ContinueLoop(Exception): pass

_brickConfiguration = {
    'wheel-diameter': 5.6,
    'track-width': 18.0,
    'actors': {
    },
    'sensors': {
    },
}
hal = Hal(_brickConfiguration)

i = 0
def run():
    i += 1

def main():
    try:
        run()
    except Exception as e:
        hal.drawText('Fehler im EV3', 0, 0)
        hal.drawText(e.__class__.__name__, 0, 1)
        hal.drawText(str(e), 0, 2)
        hal.drawText('Press any key', 0, 4)
        while not hal.isKeyPressed('any'): hal.waitFor(500)
        raise

if __name__ == "__main__":
    main()

VinArt, 2018-01-08:
Now generated code contains global token for global variables:
```
item = 0
item2 = 0
item3 = 0

def doSomething(x3, x4, x5):
    global item, item2, item3
    x5 += 1

def run():
    global item, item2, item3
    item += 1
    doSomething(0, 0, 0)
```

VinArt, 2018-01-08:
To the last "after-resolved" commit:
```python
item = None
item2 = None
def doSomething(x, x2):
    global item, item2, x
    x = None
 
def run():
    global item, item2, x
```
and before generation of "None"s was not happening, resulting in broken code.

VinArt, 2018-01-08:
last commit ment for #737

closed 2018-01-30 by bjost2s