OpenRoberta/robertalab #741: Missing type conversion ManagedString to double calliope

kwollw, 2018-01-15:
[image]
[image]

OpenRoberta with calliope permits only sending and receiving of Strings.

Collecting numeric data on the receiver and transmitting is possible due to the fact that a String can be build from numerical values. 
(transmitter: BLOCKLY: create text with get value temperature sensor < >line 20: ManagedString(ubit.thermometer.getTemperature())

On the  receiver the handling of the received data is restricted to String operations because of the lack of type conversion.
(receiver: BLOCKLY: set (numeric variable) to ... received_msg < > line 25: temp_numeric = ...)

I suggest to implement a type conversion e.g. "create number with < String" into the group Math to make numerical operations on received data possible.

As an alternative solution the sending and receiving of numbers could be implemented.

bjost2s, 2018-01-30:
We do not plan to offer a type conversion string, but we will extend the sending and receiving message block to accept numbers, see #750

bjost2s, 2018-05-07:
please reopen, if needed

closed 2018-05-07 by bjost2s