OpenRoberta/robertalab #751: Interface for interactive tutorials for the different (robot) systems

bjost2s, 2018-01-30:
As a user I would like to start programming with NEPO guided by an interactive tutorial. As the main subject of the Open Roberta Lab is programming real devices / systems, the tutorial should have steps where I can observe my system and decide by my self, if my program fulfills the task. Also variants for the simulation are possible.
Basic parts:

- choose a tutorial
- follow the instructions of e.g. step 1
- check the result / outcome 
- answer a short quiz
- go to next step
- ....
- get credentials / badges / smilies ...
- go back to the normal environment and get a hint, that you can save / export your program now

bjost2s, 2018-04-02:
The functionality is ready and we are waiting for some feedback from the designers, also real tutorials are not available. Will be reopened if the work on it will be continued

bjost2s, 2018-04-16:
Here c342b2e909ede23b4934694262afdd650f1f095f we have disabled the dir listing, so it is also forbidden on client side. The lookup in the corresponding folder has to move to the server side (init REST call).

bjost2s, 2018-04-17:
With the last commit I've also added getting available help file names, so that the client in best case does not run into the  404 (Not Found) error anymore.

rbudde, 2018-05-02:
For ev3 there is a tutorial overview available as a tutorial. This should not be deployed in the productive code. Or should it?

rbudde, 2018-05-02:
BTW: in step 3 only 2 blocks are allowed (text says 4)

bjost2s, 2018-05-03:
[image]
Blockly counts each block, not steps, so in the image abouve one can see 4 blocks, start, drive forward, math 30 and math 20.

rbudde, 2018-05-07:
Wow. This is counter-intuitive.
Tutorials ok.

closed 2018-05-07 by rbudde