OpenRoberta/robertalab #758: Chrome on Ubuntu freezes after downloading program via Run

boonto, 2018-02-02:
Browser: Chrome, Chromium
OS: Xubuntu 17.10
Robot: Calliope 2017, Microbit

After pressing the run button and downloading the generated program onto the robot the browser completely freezes. The program is executed fine on the robot.

Sometimes the browser does not freeze and the problem only occurs after the 2nd or 3rd download.

No problem when downloading to disk.

I could not reproduce this in Firefox/Ubuntu or Chrome/Windows

bjost2s, 2018-02-05:
Browser: Chrome 64.0
OS: Ubuntu 16.04 64bit
Robot: Calliope 2017

cannot reproduce this error, anyone else?

VinArt, 2018-02-05:
For me it actually freezes both on 16.04 and mint 18. I will put detailed versions later

elhele, 2018-02-06:
For me Chrome with OR freezes not only in this case. For example, it freezes when I'm trying to reload the page. 
Fedora 26, I'll check the Chrome version later.