OpenRoberta/robertalab #766: small enhancements for the C code generator

rbudde, 2018-02-09:
1. #define a constant for the sleep-time (wait blocks, ...).
2. #define a constant for 255.0/9.0, which is used at many places for calculating the brightness
3. replace the pattern if (... == true) by if(...) and if(... == false) by if(! ...). Actually the patterns are mixed (I see both (... == true) and (...)
4. replace while(true) by while(1) or vice versa. Actually we have a mix.
5. the layout of a label used in a generated goto (from a "break out from loop") is broken
6. we generate dead code after a generated goto (from a "break out from loop")
7. if a programmer chooses as function name a name used by the mbit runtime, e.g. "release_fiber", the compiler bails out. We should decide whether we ignore that or have a server side checker detecting this (will be a lot of names) or use prefixing to decrease probability of clashes or ...

These are small enhancement. May be done if there is time :-)

bjost2s, 2018-02-25:
The current implementation of #define ... will cause errors, if one declare a variable.
[image]
Please remove this part off your last commit

rbudde, 2018-03-12:
The following should be implemented:

- create a file "NEPODefs.h" and make it accessible for the code generator
- put the 3 #defines into that file. The names should start with _
- the variable uBit should be renamed to _uBit . The code generator has to be adjusted (at many places)
- the variable initTime should be renamed to _initTime and only be generated if the timer is used in the program to be compiled.