OpenRoberta/robertalab #769: Beta tester

ardu1, 2018-02-15:
Hello,

if there is a ready beta-version to use OpenRoberta with an arduino board, I would like to test it.

Thanks and best regards!
 
ardu1

elhele, 2018-02-15:
Hello @ardu1 , Botnroll is based on Arduno, so you could test some Arduino functions with it, but generic Arduino (Uno, Mega, etc.) is under development right now. It should be ready in a couple of months: https://github.com/OpenRoberta/robertalab/issues/745

ardu1, 2018-02-16:
Hello Evgeniya,

thanks for your information. Hope it will come before the end of July. 

Could help me to find the version of openrobertaUSBMacOSX? 

Thanks and best Regards

ardu1

elhele, 2018-02-16:
Hello @ardu1 ,

it must be ready by July, should be even earlier.  Even if not published to master, it will be on develop branch. I'll let you know as soon as there is something to test.

As for USB program for MacOS, I'm sorry, we forgot "MacOS" in .pkg name. Now it is fixed and you can find the file here: https://github.com/OpenRoberta/robertalab-usbprogram/releases

Best regards,
Evgeniya

ardu1, 2018-02-16:
Thanks for so fast support! :-)

ardu1, 2018-02-18:
Hello Evgeniya,

I tried this weekend to connect my Arduino UNO with Roberta. But nothing happened, only the ON LED was on and the L LED was flashing - short - long - short.

Do you have any idea?

Thanks and best regards!

ardu1

elhele, 2018-02-19:
Hello @ardu1 ,

current version of OpenRoberta can be connected only to Botnroll. However, it is not hard to change USB-program or Create-agent in such a way that it could connect to Arduino Uno. We are going to do it soon, but if you want, you can do it yourself. @VinArt  can give you a more detailed comment on this.

Cheers,
Evgeniya

VinArt, 2018-02-19:
Hi,

so, I will write something, but this implies that you can make your way through building the repository from source and starting your local server. Otherwise it is not possible to trick the server into thinking that botnroll is arduino.

WARNING! If you do the things below this line it means that if something goes wrong you might need to start over and that we are not sure 100% that we can help. Also, please, read till the end and make sure, that you want to start doing things described here.

If you are still here =) 

First follow the instructions on building the repo from source. You will need git, maven and JDK. The instructions themselves are in README in this repo.

Second. Locate the file robertalab/OpenRobertaParent/RobotArdu/src/main/resources/botnroll.properties, you need to edit it and change robot.vendor = 0x10c4 to robot.vendor = ARDUINO_UNO_VID, where ARDUINO_UNO_VID can be one of the following: 0x2341, 0x2A03 (maybe something else for other arduinos, not UNO, I mean). I am not sure which vendor ID your arduino has, so you need to figure that one on your own. For example in Linux you can type `lsusb` while Arduino is connected and check the ID parameter, I beleive you need to use the first part, before the colon.

Once you have edited the file, recompile the project, for safety. You can use -DskipTests parameter for maven, to speed up the process.

Run the server.

Third thing to do would be to get create-agent from here: https://github.com/OpenRoberta/create-agent/releases/tag/untagged-810957d91a3531b3b045, install and run it. Do pay attention that you will have to edit the create-agent configuration, since it is shipped with default one that works with on-line lab. Go to where the agent was installed, on Linux it is $HOME/ORCreateAgent and locate the config.ini. Please note, that if your are under Windows, use Notepad++ to edit this file, otherwise line endings will get corrupt. You will need to change origins from origins = https://lab.open-roberta.org to origins = http://localhost:1999.

Save and run the agent.

Now when you go to http://localhost:1999 you should be able to select bot'n'roll and your arduino will be detected as this robot.

If you don't want or cannot for some reason use create-agent, you can actually clone robertalab-usbprogram repository and edit the following file: robertalab-usbprogram/OpenRobertaUSB/src/main/java/de/fhg/iais/roberta/connection/ArduUSBConnector.java
what you would need to change is  `private boolean findArduinoLinux()` method and replace `Silicon_Labs_CP2104_USB_to_UART_Bridge_Controller` with probably (!) `Arduino` but it is better to check the `/dev/serial/by-id/` directory. If you are running windows replace the "string Silicon Labs" in `private boolean findArduWindows()` with also, maybe "Arduino"
Rebuild the usb-program and give it a try. While I have tried doing this for create-agent, I never tried it with usb-program, so no idea how it will work out.

Side note. The above process should work for any ATMega 328p based board as long as you get the right VID in the properties file.

Another warning. While this _should_ connect OpenRoberta with Arduino, bot'n'roll blocks are not _meant_ to be used with an Arduino, so don't expect things like show text work on a shield or block lights to turn on and off. I would say that only generic programming blocks will do something meaningful, and since currently there is no possibility to do Serial.print, you won't be able to get the data out of Arduino. Also pin configuration is set for bot'n'roll, so if you plug an LED where it is not expected by bot'n'roll you may burn some hardware, though it is unlikely. I think the most that you will get out of this procedure is to see arduino being flashed.

closed 2019-06-26 by rbudde