OpenRoberta/robertalab #770: Random number generation in BlocklyMethods is broken

VinArt, 2018-02-16:
# issue
file blockly_methods.py in class BlocklyMethods has the following method for generating random numbers:
```python
    # note: we don't use the random module since it is large
    @staticmethod
    def randInt(min_val, max_val):
        b = os.urandom(4)
        val = b[0] << 24 | b[1] << 16 | b[2] << 8 | b[3]
        if min_val < max_val:
            return min_val + (val % ((max_val - min_val) + 1))
        else:
            return max_val + (val % ((min_val - max_val) + 1))
```

unfortunately os.urandom returns a string and it cannot be cast to int for following shift operations. It results in the following exception being thrown:

```
robot@ev3dev:~$ python
Python 2.7.9 (default, Aug 13 2016, 17:33:18) 
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> b = os.urandom(4)
>>> val = b[0] << 24 | b[1] << 16 | b[2] << 8 | b[3]
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: unsupported operand type(s) for <<: 'str' and 'int'
```

As per python [documentation](https://docs.python.org/2/library/os.html) indeed a string is returned:

```
os.urandom(n)
Return a string of n random bytes suitable for cryptographic use.
```
# possible solution
Probably the solution would be to do something like this:

```python
val = int(os.urandom(4).encode('hex'), 16)
```

since os.urandom returns a string of bytes, for example (good example, everything is out of printable range, so printed as hex numbers): 

```
In [4]: os.urandom(4)
Out[4]: '\xad\xac\xf5\xa2'
```

we can take that, encode as hexadecimal numbers and construct an int from base 16 and use it further.

A test program on ev3 that generates 10000 (10k) numbers runs for ~34-35 seconds with printing out the number to console. Without print-out ~6-7 seconds, so this conversion should not slow down user programs too much. NAO should be affected even less, since it has a better CPU (Atom z530).

VinArt, 2018-02-16:
For ev3dev in order to fix randDouble the following can be done:

```python
return 1.0*int(os.urandom(4).encode('hex'), 16) / 0xffffffff
```

ensonic, 2018-02-23:
Oh, this is fallout from the python2-3 port. FYI, the reason we're not using the regullar random module is the slowness of importing it :/
Your solution looks good to me. Thanks for investigating. Could you add a test covering it?

VinArt, 2018-02-23:
Sure, no problem. I will then fix this for both NAO and EV3.

VinArt, 2018-02-28:
So I tested microbit as well, it uses random module. I don't know whether it's CPU is much better or there are other reasons, but the 10k integer test takes 1,2 seconds and 100k integer test takes 12 seconds. And it seems that it does not take so long to load.

VinArt, 2018-03-02:
Also, sorry for the first proposal, it works only in python2, for python3 there is a different solution (int.from_bytes(os.urandom(4), byteorder='big')).

bjost2s, 2018-05-07:
to which project does this issue belong?

rbudde, 2018-09-03:
for NAO + Vorwerk the problem is solved.

But I don't see a fix for ev3dev in `robertalab-ev3dev/roberta/BlocklyMethods.py`. Please comment on this.

closed 2019-02-22 by rbudde