OpenRoberta/robertalab #773: inconsistent type constraints calliope enhancement

kwollw, 2018-02-26:
[image]

**"show text" can have both String and Number arguments.**
example block code translation:
uBit.display.scroll(ManagedString("Hallo"));
uBit.display.scroll(ManagedString(42));  <<<implicit type conversion possible

**"send message String" can only have String arguments.**
example block code translation:
uBit.radio.datagram.send(ManagedString("Hallo"));
uBit.radio.datagram.send("");   <<< cannot insert "ManagedString(42)" here

Both cases should be treated equally: If there is an implicit type conversion Number -> String it should apply in all cases where a String argument is required.

There are other cases with implicit type conversion or strict type restriction.

I would appreciate strict type control and type conversion functions like  to_s and to_i

bjost2s, 2018-02-26:
we are already working on this, please see #750, will be released soon

kwollw, 2018-02-26:
The issue here is to focus on consistency in type handling. **send message String** vs. **show text** is only an example for a non consistent type concept.
For educational reasons there should be a rule throughout the language: Either strict type constraints or implicit type conversion. In NEPO it is mixed.

rbudde, 2019-06-26:
we're working on typechecking NEPO (see expression block branch). I agree, that we should have a fix rule. I'd prefer constraints :-)

closed 2019-06-26 by rbudde