OpenRoberta/robertalab #774: mvn clean install on branch master exits with error

kwollw, 2018-02-26:
mvn clean install on branch master exits with error (RobotArdu failed)
**Running de.fhg.iais.roberta.ast.action.MotionParamTest** 
(3 Errors)

bjost2s, 2018-02-26:
I cannot reproduce this error, please try
`mvn clean install -U`
If this is still not working, please clone the repo again and run 
`mvn clean install`

VinArt, 2018-02-28:
I just wonder.. Are you using AMD or Intel CPU?

kwollw, 2018-02-28:
Intel(R) Core(TM) i5-3210M CPU @ 2.50GHz, 64 bit

VinArt, 2018-02-28:
It was off-topic. So maven executes tests non-deterministically and the order may vary from platform to platform. We have an issue of dependency between different tests that we are solving now. So I asked because I wonder what affects the order of execution.

bjost2s, 2018-03-01:
Please use `mvn clean install -DskipTests`  to build the project until we found a solution for this.

rbudde, 2018-03-01:
I can reproduce your problem.
As Artem already wrote, we have (unwanted) dependencies between tests (due to a static variable :-<). I'll scan the tests for this erroneous pattern + remove the occurences till Tuesday. Until this has been done, pls skip the tests as Beate proposed.

Thank you very much for your ticket!

rbudde, 2018-03-01:
I committed bf497ddf8f929d6cbb45e931192448639c52ed51 in branch develop. You'll get the changes if you pull develop. Can you check whether the error is gone? On my machine everything is fine now.

kwollw, 2018-03-01:
I confirm that mvn clean install on branch develop makes build successful on my machine too.

rbudde, 2018-03-09:
see #336 (the same bug)

rbudde, 2018-03-12:
Solved

closed 2018-04-05 by bjost2s