OpenRoberta/robertalab #777: Text 2 Speech Ipad Simulation

Thorlogy, 2018-03-01:
The Text2Speed Block is not working in the Simulatoin on Ipad 
Tested with Chrome and Safari - latest OS version

boonto, 2018-03-02:
According to [Browser compatibility](https://developer.mozilla.org/en-US/docs/Web/API/SpeechSynthesis#Browser_compatibility) and [Browser support](https://developers.google.com/web/updates/2014/01/Web-apps-that-talk-Introduction-to-the-Speech-Synthesis-API?hl=en#browser_support) iOS/Mobile Safari 7.1 is partially supported. [Here](https://developer.apple.com/documentation/webkitjs/speechsynthesis) it says Safari Mobile 9.0+, so I guess it should work on the newest version.

Maybe its related to the soft mute functionality similar to [this issue](https://forums.developer.apple.com/message/293874#293874).

I can not test it right now, because I do not have an IPad.

bjost2s, 2018-03-02:
not sure if Ipads allow this in general, so maybe it can't be fixed

bjost2s, 2018-03-02:
@Thorlogy could you please check the settings provided in the similar issue @boonto has provided?

boonto, 2018-03-05:
I found [this](https://stackoverflow.com/questions/32193704/js-speech-synthesis-issue-on-ios), apparently the user has to trigger the action, pure javascript will not work.

rbudde, 2018-03-12:
To be reviewed later. Can't be fixed with the actual IOS versions due to security issues.

closed 2019-06-26 by rbudde