OpenRoberta/robertalab #780: C/C++ function code generation crashes if return value not specified

VinArt, 2018-03-02:
Running this kind of a program:
[image]
results in "This program cannot be transformed into machine code". While this is true and is obvious why, since function must return a value, on larger programs it might be hard to figure out that the return value was omitted somewhere without explicit error message.

rishabhdeepsingh, 2018-03-02:
Will return 0 in main function will solve this issue?

VinArt, 2018-03-02:
@rishabhdeepsingh the issue lies within doSomething() function, since return keyword is used without return value. In main the whole return is omitted, which is fine.

rishabhdeepsingh, 2018-03-02:
What changes should we make then?

tanhengyeow, 2018-03-02:
Would enforcing all functions returning the declared type fix this issue?

E.g.
`return 0.0` for double
`return 0` for int
`return null` for String

VinArt, 2018-03-02:
@tanhengyeow yep, something like that. It is somewhere in the core.

lccopp, 2018-03-09:
Maybe we could set the initial value to null, and set a button to let user add return value.

It means merge these two modules into one.
[image]
and it will turn to this
[image]
when click "+" at bottom, then we can add a return value.

bjost2s, 2018-03-10:
Our blocks are already used by thousands of users in even more programs, so changing blocks is more or less impossible and not wanted.
Also returning 0 may confuse the user, because if you simply forgot to add the return value it is even hard to find the error at runtime if 0 is returned.
Maybe consider the compiler output.

rbudde, 2018-03-12:
A program that is not complete, should not be completed by our system. Thus, if a user has forgotten the return value, this should be detected by our system (when checking or generating code, both for simulation and a real robot), send back to the front end and an annotation should be added to the block to inform the user about his/her error.
I believe, this is the only nice solution, but not an easy task. Anyboby may contact @bjost2s or @rbudde

boonto, 2018-03-16:
Merged from #792 
[image]

All of the non comment blocks in the image crash and lead to "This program cannot be transformed into machine code" errors (only with missing parameters, of course). All of these should be caught by the code checking and give students better feedback of whats wrong with their program.

These are just the EV3 blocks, for the other robots this should also be the case.

rbudde, 2018-03-19:
I agree.

boonto, 2018-03-21:
Currently we implicitly add default values if a parameter is missing. This should be changed and also covered by the checker in the future.
[image]