OpenRoberta/robertalab #782: Color sensor detection boundary is not clear

lccopp, 2018-03-08:
[image]
In this case, there is no grey, but the gray was detected.
[image]
In this case the ground line is black, but I need to use grey to make it run.

bjost2s, 2018-03-10:
Sorry for this. Grey is not grey but no colour. Unfortunately there is no way to label it differently then grey. The simulation is here close to the real color sensor which is returning a colour or if no colour could be detected, none. In your example the sensor is over yellow and white which results in no colour.

For your project, line following, you should use the light mode from the colour sensor which is returning grey values from 0 (dark) to 100 (bright).

bjost2s, 2018-04-12:
Please reopen, if you have further questions

closed 2018-04-12 by bjost2s