OpenRoberta/robertalab #789: AdHoc groups for group work on the same server

ensonic, 2018-03-16:
It would be nice to have a way to associate sessions on the server into a group. The is key to enable group work where robots collaborate (including robot to robot messaging).
The whole design would not require accounts to make it easy to use. I imagine this workflow:

1. one user selects from the menu `Groups/New group ...`
   1. a window shows up where the user can name the group, name himself (if guest) and click okay
   1. this creates the group and will show a window with the group pairing code (and the people that have joined already)
1. now other users on the same server can go to the menu and select `Groups/Join group ...`
   1. a window will pop up where they must enter the pairing code and can change their name (if guest)
1. for each joined guest the name will show up in the window for the group creator
1. the group creator has now a menu enabled that contains:
   1. 'Groups/Close group ...', this will ask for confirmation and remove the group with all the joined people.
   1.  `Groups/Info ...`. this will re-show the window with the pairing-code and the number of people that are conencted.
1. users that have joined, have a menu `Groups/Leave ...' that after confirming removed them from the group

With this setup we can implement the ideas below. If this sounds good, we'll create sub-tickets for each of them later.
* [ ] resolve bluetooth addresses - we add the mac addr of the bt interface to the robotinfo. The server can build a Map of name to mac addr and include this into generated code. The code can use the Map to go from name to mac-addr. We might even beable to tweak the blockly blocks, so that one can pick the name of the robot from a drop-down (more work?)
* [ ] share programs with the group
* [ ] collaborative simulation - this would allow to also add other group members robots to the simulation.

rbudde, 2018-03-16:
I like this proposal. But we need to investigate:

- sharing: actually sharing is restricted to programs stored in the db. This in turn expects login. But we could store a map groupId -> (user,programName, program) temporarily (maybe even in memory?) to be accessable for loading into the browser by other group members.

- bluetooth: user like to save a blockly program as xml and reload it later. The BT naming mechanism may introduce a block "connect to robot" with two params group and name, instead of one parameter. Furthermore a user may save and load native code (in a version not far in the future :-)). Code generation must be robust w.r.t. to groups (not available anymore).

The workflow 1-5 is fine for me.

elhele, 2018-03-16:
We could may be combine it with the dashboard. Now it looks like this (issue #384, #700, #702-#706):

-- user can go to groups from the menu:
[image]
-- create/delete groups in a group tab:
[image]
-- add/remove users in a group's user tab:
[image]

It could probably just easier to add a buttons "create an AdHoc group" and "join AdHoc group", something like this (I don't mean the design of the buttons, just position :-)):
[image]

And keep the rest of workflow how it is described in 1-5.

ensonic, 2018-03-16:
Thanks for sharing the images of the groups ui. The workflow can indeed become simpler if we stay in a single window. I can see the challenges in the icon design though and potential confusion with users. If uses are not logged in we can of course gray out the `[+]` button and tell them that only logged in users can create a persistent group.
For the persistent groups, is it so that the owner adds people?

elhele, 2018-03-19:
Ok, it makes sense. Then I should probably change it for permanent groups as well, so it would be uniform.
Yes, now the owner adds people, however, it can be a good idea to introduce the token for permanent groups as well. It is a great solution for spam problem -- now a group owner can just "spam" other users by adding then to the group, but if the users will be adding themselves by token, we avoid this problem. Also, the teacher wouldn't have to input all the student's names, they'll just have to input the token.
The only problem is that the idea was that the teacher can also create new users for the group, but it can be a separate feature.