OpenRoberta/robertalab #790: horizontal scaling with docker

rbudde, 2018-03-16:
in directory Docker there is a README describing how openroberta is put into docker container. There is also a description how to create these containers with docker, but that is not interesting here. The standard docker deployment is done with 2 container using docker-compose file dc-server-db-server.yml

- one container is the database server
- the other container contains the server and all crosscompiler

Calling the crosscompilers is likely the resource-intensive part. Thus the following idea arises:
- create a third container containing all the cross compiler (the dockerfile for that is already there, because it is used for creating the server container)
- deploy this container as often as needed

Now the "challenge": we need to add **load balancer support** to our (main) server. Standard load balancer won't work (pls convince me from the opposite ...). This is needed because we have to bundle 2 tcp-connections from different clients (robot + browser) and have to guarantee, that the program from the browser is sent to the crosscompiler container, that is used by the robot to retrieve the binary from. This is a non-standard kind of stickyness for a load balancer.

I propose first to assemble ideas, before we start any imlementation :-)

VinArt, 2018-03-16:
I think, that as long as we have 1 container with the openroberta server we can deploy as many compiler containers as we want and do balancing there. When we exceed the throughput on the server container, then we will have to think of load balancing incoming connections.

If you mean, that we need many server containers, then we have to make this custom load balancer.

rbudde, 2018-03-19:
the flow (with 1 ora server) is as follows for robots as ev3:

- browser request to run a program arrives at the ora server. Server either takes the program/configuration or loads it from the database and sends it to a compiler container
- the compiler container creates the binary and saves it to disc. The compiler container sends a response to the ora server
- the ora server does two things:
  - sends a response to the user ("... program will execute soon ...")
  - sends a response to the robot (that is connected using the token) to trigger a download of the compiled program
- the robot sends a download request. This must be redirected to the compiler container

This is what a meant with a kind of "sticky load balancing": the ora server has to select the compiler container not only for the browser (this would only need a simple redirect), but has to remember that compiler container also for the robot (which uses a different connection)

For Calliope the things are a lot nicer:
- the ora server can immediately redirect to a compiler container
- the compiler container will return the binary as part of its response

Here we have to think about race conditions between browser/ora server/compiler container.