OpenRoberta/robertalab #792: Missing parameters lead to compiler crashes

boonto, 2018-03-16:
After #780 I looked into other blocks with similar behaviour.

[image]

All of the non comment blocks in the image crash and lead to "This program cannot be transformed into machine code" errors (only with missing parameters, of course). All of these should be caught by the code checking and give students better feedback of whats wrong with their program.

These are just the EV3 blocks, for the other robots this should also be the case.

bjost2s, 2018-03-16:
As for #780, this is not a bug. It is the current way how we check the code. Again, the idea is to either check the compiler error message and derive from there detailed information to inform the user or to improve the checker from the codegeneration.
Most of them are not robot specific, so no need to create a ticket for each system. Please close one of the two and only use one in the future.

boonto, 2018-03-16:
Most of the discussion was in the other one so I will close this one and change the title of #780 to be more generic.

closed 2018-03-16 by boonto